drandyhaas/Haasoscope #3: OLED doesn't show anything past boot screen

tedder, 2018-05-01:
See picture for my OLED install. It's on the leftmost pins, leaving three pins on the right unconnected. That appears to match what's in the documentation.

[image]

It shows that (slightly odd) boot image and doesn't change. Tried powering off and back on, no change.

drandyhaas, 2018-05-01:
Once you readout an event, it should show that event (from the selected channel). Are you reading out events?

tedder, 2018-05-01:
I haven't connected it (still installing Qt dependencies). I assumed it would work when standalone.

drandyhaas, 2018-05-01:
No, it does nothing until you read out events. I'll clarify this in the guide.

closed 2018-05-01 by drandyhaas