drandyhaas/Haasoscope #6: Program doesn't start on Mac

salmansheikh, 2018-05-05:
I tried both python and python3 on the program in the terminal app on OSX and it gives errors. 

python HaasoscopeDraw.py 
  File "HaasoscopeDraw.py", line 40
    if d.db: print time.time()-d.oldtime,"done with evt",nevents
                      ^
SyntaxError: invalid syntax

GSSLA16111692:software ssheikh$ python3 HaasoscopeDraw.py 
  File "HaasoscopeDraw.py", line 40
    if d.db: print time.time()-d.oldtime,"done with evt",nevents
                      ^
SyntaxError: invalid syntax

tedder, 2018-05-05:
`python` is linked to `python3`. You can hit `python --version` to verify.

Use `python2` instead.

drandyhaas, 2018-05-05:
Did that solve your problem?

salmansheikh, 2018-05-05:
Nope. I tried it on my iMac and it had different errors. First no pyserial,
which I installed and thought its Python 2.7.10 (OSX 10.13.4) by default on
when I run python -V.

I am getting this error that I don't understand how to fix:

python HaasoscopeDraw.py
Loading HaasoscopeLib.py
Traceback (most recent call last):
  File "HaasoscopeDraw.py", line 1, in <module>
    import HaasoscopeLib
  File "/Users/ssheikh/Haasoscope/software/HaasoscopeLib.py", line 18, in
<module>
    import matplotlib.pyplot as plt
  File
"/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/pyplot.py",
line 98, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show =
pylab_setup()
  File
"/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/__init__.py",
line 28, in pylab_setup
    globals(),locals(),[backend_name],0)
  File
"/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/backend_qt4agg.py",
line 13, in <module>
    from backend_qt4 import QtCore, QtGui, FigureManagerQT, FigureCanvasQT,\
  File
"/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/backend_qt4.py",
line 25, in <module>
    from qt4_compat import QtCore, QtGui, _getSaveFileName, __version__
  File
"/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/qt4_compat.py",
line 36, in <module>
    import sip
ImportError: No module named sip

tedder, 2018-05-05:
Those errors show you got past using py3 like you were in the first errors. If it's failing at sip you haven't followed the guide :)

salmansheikh, 2018-05-15:
But I never got it working yet...I will try on windows. I just bought a
license for a Virtualbox Windows VM. Hopefully it works in there..

PeterAP, 2018-11-04:
Well - not possible to start software. Where the Haasoscope.py must be located?
I use Mojave, Python2.7 - all seems ok. 
Bui on Terminal I get:
MacBookAir:site-packages petera$ python2 HaasoscopeDraw.py
/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python: can't open file 'HaasoscopeDraw.py': [Errno 2] No such file or directory

drandyhaas, 2018-11-04:
Hi,
You should be in the Haasoscope/software directory to run HaasoscopeDraw. Please see the guide:
https://goo.gl/1Ry1k4

PeterAP, 2018-11-04:
Yes it is - but where to place tha Haasoccope folder? I have it on Desktop - Terminal says "no such file"
I clone it with GitHub app and then it was in Document folder but result is the same...
I clone with GiHub.app and Haasoscope folder are in Document folder... May be full path must be written?

PeterAP, 2018-11-04:
Got it! More or less full path must be shown: 
cd Documents/GitHub/Haasoscope/software
And it goes on! Thanks!

PeterAP, 2020-03-12:
Hello!

We communicate a while ago and I use Haasoscope time after time.
Now I switch to other computer and cannot get it working again… something doesnt work when I follow the User guide.

Here I send Terminal screenshoot.

Some ideas?

Thanks!

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

salmansheikh, 2020-03-12:
I didn't see a terminal shot. Perhaps it got blocked?

PeterAP, 2020-03-12:
Hmmm…must be in attachment…
Here I resend.
Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

PeterAP, 2020-03-12:
Here it is as a copied text from Terminal:
Last login: Thu Mar 12 22:48:54 on ttys000
MacBook-Air:~ petera$ cd Desktop/Haasoscope/software
MacBook-Air:software petera$ python HaasoscopeDraw.py
Loading HaasoscopeLib.py
('Darwin', 'MacBook-Air.local', '18.7.0', 'Darwin Kernel Version 18.7.0: Sun Dec  1 18:59:03 PST 2019; root:xnu-4903.278.19~1/RELEASE_X86_64', 'x86_64')
matplotlib backend is MacOSX
Loading HaasoscopeLib.py
('Darwin', 'MacBook-Air.local', '18.7.0', 'Darwin Kernel Version 18.7.0: Sun Dec  1 18:59:03 PST 2019; root:xnu-4903.278.19~1/RELEASE_X86_64', 'x86_64')
matplotlib backend is MacOSX
num main ADC and max10adc bytes for all boards =  2048 and 0
rate theoretically 35.95 Hz over serial
/dev/cu.wchusbserial1410 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=20-1
/dev/cu.usbserial-1410 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=20-1
/dev/cu.Bluetooth-Incoming-Port : n/a : n/a
connected serial to /dev/cu.wchusbserial1410 , timeout 3.0 seconds
got firmwareversion 0 for board 0 in 104.4 ms
minimum firmwareversion of all boards is 0
rolling trigger
num samples is 512
send increment is 0
do highres is True
trigger time over/under thresh now 1 and usedownsamplefortriggertot is True
serialdelaytimerwait is 100
tell SPI setup: 131  8 0
tell SPI setup: 131  6 16
tell SPI setup: 131  4 36
tell SPI setup: 131  5 36
tell SPI setup: 131  1 0
usb2 connection not available
got uniqueID 80988211106e3f28 for board 0 , len is now 1
current directory is /Users/petera/Desktop/Haasoscope/software
reading calibrations for board 0 , channels 0 - 4
read calib/calib_80988211106e3f28.json.txt
calib was written using unknown firmware version
do highres is False
shut down adcs
bye bye!
Traceback (most recent call last):
  File "HaasoscopeDraw.py", line 73, in <module>
    d.redraw()
  File "/Users/petera/Desktop/Haasoscope/software/HaasoscopeLib.py", line 1436, in redraw
    self.figure.canvas.flush_events()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backend_bases.py", line 2360, in flush_events
    raise NotImplementedError
NotImplementedError
MacBook-Air:software petera$

Kind regards,
Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

salmansheikh, 2020-03-12:
This sounds like a Dr Haas type of question. I know more about h/w then
software as I’m an EE mostly doing FPGAs. I did do some work with RISC-V
but I struggled with the sw toolchain.

PeterAP, 2020-03-13:
Hi!
I see.
Can you pas the question to Dr. Haas, pleas?

Kind regards,
Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

salmansheikh, 2020-03-13:
I think he should see this anyhow by you posting it. I don’t have any
direct email for him. You could check NYUs website for his info.

drandyhaas, 2020-03-13:
I do see all these comments. I'm not a Mac user myself, so it's hard to debug. You could try commenting or the offending line 1436. You could also try the new Qt version. HaasoscopeQt. And maybe someone else with a Mac can help you.

salmansheikh, 2020-03-13:
I have a iMac and can try later. I did get it to start up on it a while
back but its my home office and not conducive to electronics work. I built
a Ryzen Linux system for my shop for that. But I can retry later today or
tomorrow on the iMac. Its a 2010 but I have the latest OS. I remember it
took a while to get it working.

PeterAP, 2020-03-13:
Hello!
Thanks!

Kind regards,
Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

PeterAP, 2020-03-13:
Hi!
After opt-out lines
  def redraw(self):
        if self.domaindrawing: # don't draw if we're going for speed!
            if dofast:
                self.ax.draw_artist(self.ax.patch)
                if self.dogrid:
                    [self.ax.draw_artist(gl) for gl in self.ax.xaxis.get_gridlines()]
                    [self.ax.draw_artist(gl) for gl in self.ax.yaxis.get_gridlines()]
                if self.needtoredrawtext: [self.ax.draw_artist(l) for l in self.texts]
                self.needtoredrawtext=False
                [self.ax.draw_artist(l) for l in self.lines]
                [self.ax.draw_artist(l) for l in self.otherlines]
                self.figure.canvas.update() #needs Qt4Agg backend
            else:
                self.ax.relim()
                self.ax.autoscale_view()
                self.figure.canvas.draw()
       #  if len(plt.get_fignums())>0:
       #      self.figure.canvas.flush_events()
    I get some calibration?:
got uniqueID 80988211106e3f28 for board 0 , len is now 1
current directory is /Users/petera/Desktop/Haasoscope/software
reading calibrations for board 0 , channels 0 - 4
read calib/calib_80988211106e3f28.json.txt
calib was written using unknown firmware version
100 events, 5.01 Hz
200 events, 5.0 Hz
300 events, 5.01 Hz
400 events, 5.0 Hz
500 events, 4.99 Hz
getdata asked for 2048 bytes and got 1991 from board 0
(110, 111, 111, 110, 111, 111, 111, 111, 111, 111)
600 events, 4.39 Hz
700 events, 5.0 Hz
800 events, 5.0 Hz
getdata asked for 2048 bytes and got 2003 from board 0
(111, 111, 111, 111, 110, 111, 111, 110, 110, 110)
900 events, 4.38 Hz
getdata asked for 2048 bytes and got 2016 from board 0
(111, 111, 111, 110, 110, 110, 111, 110, 110, 110)
1000 events, 4.39 Hz
1100 events, 5.0 Hz
getdata asked for 2048 bytes and got 2038 from board 0
(110, 110, 110, 110, 110, 110, 110, 110, 110, 110)
1200 events, 4.38 Hz

And it continues for hours…

Andy ideas?

Kind regards,

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

drandyhaas, 2020-03-13:
Please see the troubleshooting section of the Haasoscope guide.

If you constantly get errors printed in the terminal like "getdata asked for 16384 bytes and got 7904 from board 0", your USB port is likely not up to snuff. Yes, the data rate from serial, even at 1.5 Mb/s is pretty small even for "full speed" USB2, which should be capable of 12 Mb/s. But the buffering is sometimes an issue, leading to packet loss. The best is if you plug Haasoscope into a USB 3.0 port. If your computer doesn't have USB 3.0 you can easily add it. Also don't use a USB hub if you have problems - plug into your computer directly! 
Using the optional USB2 Haasoscope module will also solve the problem.
A workaround is to set the "serialdelaytimerwait" variable to 600 (or whatever works for you!) in the HaasoscopeDraw.py script, to slow down the serial writing from the board. This is roughly how many clock ticks to wait before sending each byte, so the smaller the number the faster the serial data rate. 
If your computer has a built in serial port, like the raspberry pi GPIOs, you can also use that directly (see raspberry pi zero w instructions above).

PeterAP, 2020-03-14:
Hello!

Well - I change "serialdelaytimerwait" variable to 600. No messages like 
"getdata asked for 2048 bytes and got 1991 from board 0
(110, 111, 111, 110, 111, 111, 111, 111, 111, 111)”
anymore, but still stuck in the loop “24000 events, 5.0 Hz”…

Originally MacBook Air has two USB 3.0 ports, and I have tried both of them...

Regards,
Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

drandyhaas, 2020-03-14:
Hmm. You mean the screen doesn't update?
Maybe try the new Qt gui. It's better!

PeterAP, 2020-03-14:
I cannot get a screen at all.
How to try Qt?

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

drandyhaas, 2020-03-14:
python HaasoscopeQt.py

PeterAP, 2020-03-14:
Can’t get it working:

MacBook-Air:~ petera$ cd Desktop/Haasoscope/software
MacBook-Air:software petera$ python HaasoscopeQt.py
Traceback (most recent call last):
  File "HaasoscopeQt.py", line 8, in <module>
    from pyqtgraph.Qt import QtCore, QtGui
ImportError: No module named pyqtgraph.Qt
MacBook-Air:software petera$ 

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

> On 14 Mar 2020, at 13:26, Andy <notifications@github.com> wrote:
> 
> python HaasoscopeQt.py

drandyhaas, 2020-03-14:
Try
pip install pyqtgraph

PeterAP, 2020-03-14:
Tried:

Successfully built pyqtgraph
Installing collected packages: pyqtgraph
Successfully installed pyqtgraph-0.10.0
MacBook-Air:software petera$ cd Desktop/Haasoscope/software
-bash: cd: Desktop/Haasoscope/software: No such file or directory
MacBook-Air:software petera$ python HaasoscopeQt.py
Traceback (most recent call last):
  File "HaasoscopeQt.py", line 8, in <module>
    from pyqtgraph.Qt import QtCore, QtGui
  File "/Users/petera/Library/Python/2.7/lib/python/site-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui
  File "/Users/petera/Library/Python/2.7/lib/python/site-packages/pyqtgraph/Qt.py", line 47, in <module>
    from PySide import QtGui, QtCore, QtOpenGL, QtSvg
ImportError: dlopen(/Users/petera/Library/Python/2.7/lib/python/site-packages/PySide/QtGui.so, 2): Library not loaded: @rpath/libpyside-python2.7.1.2.dylib
  Referenced from: /Users/petera/Library/Python/2.7/lib/python/site-packages/PySide/QtGui.so
  Reason: image not found

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

drandyhaas, 2020-03-14:
https://github.com/pyside/PySide/issues/154
Maybe that helps?

PeterAP, 2020-03-14:
> pyside/PySide#154 <https://github.com/pyside/PySide/issues/154>

No, I try different but nothing help.

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

PeterAP, 2020-03-14:
Just a question:
is it possible to find out how the commands are structured in serial_read.py?
For example:

 [124, 3], #downsample 3
 [125, 1], #tickstowait 1
 [136, 2, 32,  0,   0, 255, 200], # io expanders on (!)

what is the meaning for [136, 2, 32,  0,   0, 255, 200] for example>

Kind regards,

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

drandyhaas, 2020-03-14:
Yes, see "General communication and readout philosophy" section in the guide.
It references this file:
https://github.com/drandyhaas/Haasoscope/blob/master/software/commands.md

PeterAP, 2020-03-14:
Ohhh…I was “blind”in a bit..
Thanks!

Dr.Sc.ing. Peteris Apse-Apsitis

Phone: +371 29184559
E-mail: peter.apse@gmail.com

closed 2021-04-17 by drandyhaas