drandyhaas/Haasoscope #8: Windows - ImportError: No module named serial

aalku, 2018-05-08:
```
C:\Haasoscope-master\software>python HaasoscopeDraw.py
Loading HaasoscopeLib.py
Traceback (most recent call last):
  File "HaasoscopeDraw.py", line 1, in <module>
    import HaasoscopeLib
  File "C:\Users\alberto\Desktop\Haasoscope-master\software\HaasoscopeLib.py", line 11, in <module>
    from serial import Serial, SerialException
ImportError: No module named serial
```

I guess I need a windows version of that module or maybe the software should use a serial library that supports Windows. I don't know.

Please help.

drandyhaas, 2018-05-08:
Yes, you need the pyserial library. Did you install all the required libraries mentioned in the Haasoscope guide?
"pyserial, numpy, matplotlib, scipy, and maybe pyside"

aalku, 2018-05-08:
I didn't find windows install documentation and didn't carefully read linux install documentation. Sorry. I'll try to find out how to install that modules on windows. Thank you.

aalku, 2018-05-08:
It's working!
Install steps for windows were:
1. Install Python 2.7
2. Run this command line: `C:\Python27\Scripts\pip.exe install pyserial numpy matplotlib PySide Qt.py scipy`
3. Enjoy!

drandyhaas, 2018-05-08:
Thanks! Added to guide..