drandyhaas/Haasoscope #10: Probes won't compensate

g8sqh, 2018-05-25:
Cut down the probe trim tool and adjusted the pcb trimmers to get nice square waves

The probes supplied won't compensate. This is fixable thus:
- Pull the cable strain relief away from the BNC plug, and slide off the small 'box' cover
- Next to the black trimmer on the probe PCB solder on a 120 (one hundred and twenty) pF capacitor onto the two empty pads
- Refit the cover and the strain relief

This done the probes will also compensate (normal gain on PCB).

Thanks for a great product

drandyhaas, 2018-05-25:
This is true. And thanks for the tip/fix!
To be clear for others, the probe compensation (screw on the probe itself) only affects the 10x probe setting (switch on the probe itself). 
Also, the lack of compensation only affects signals below ~50 kHz, which is why I never worried too much about it. I think the reason it fails to compensate with the default cap in the probe is that I have too much (or too little?) capacitance at the BNC input. (?)

g8sqh, 2018-05-25:
I don't understand the value I needed. A normal 'real' old-fashioned scope was a fixed 30pF in parallel with 1M - they actually had trimmers on the input attenuator to keep the value constant (as well as another trimmer for compensation)

Later scopes went down to 20pF, and I think a few reached 14pF - always to reduce the loading on the circuit being measured. The screw in the probe box is directly across the BNC connector - and thus corrects for the 14-30pF range of real scopes.

HOWEVER - I had to add 120pF - not a typo - took 4 tries adding more to get it right. This tells me that the probe tip end is probably made wrong - I suspect they may have some stray capacitance across the 9M series resistor which they'd not designed for. Your BNC input would have to be around -100pF if the probe was made right - and I know of few circuits that produce a wideband negative capacitance

I'll measure the input capacitance when I have a chance - but certainly don't add capacitance to the scope PCB BNC input. I also have some known working probes on my analog scope - will try them on the Haasoscope.

Thanks for designing this PCB - I have in the past had to video my analog scope to capture rare events - digital storage is really nice to have. 

Best wishes

--David

drandyhaas, 2018-05-25:
Ahh. I think it's not the input capacitance that's the issue, actually. I played with adjusting it from 7-15 pF (it's nominally 10 pF), and not much changed. You'll also notice that the quick AC part of the response is of the correct magnitude (10x smaller than 1x), but the DC part of the response is too small. I think the input _resistance_ is too low. It's supposed to be 1 MOhm, but when it's in parallel with the input impedance of the op-amp, it drops to 600 kOhm. 

So really this 1 MOhm resistor on the board, pointed to in the pic here, should be ~10 MOhm. Seems to work OK, see below (that's using the probe at 10x setting).

[image]

[image]

g8sqh, 2018-05-25:
I've managed to open the schematic, and you're right

On normal gain the input impedance is 500K - the 1M to ground (e.g. RIN 5) in parallel with the 1M to the opamp input (e.g. RIN1). (The opamp is wired inverting, so the inverting input is a virtual ground at low frequencies. The differential impedance is 60k on the datasheet - so still 515k at higher frequencies) 

The 120pF now makes sense - we have the capacitance of about 1m of cable plus the probe trimmer + 14-30pF of scope. Probe cable is special, but I suspect 100pF/m isn't far off, probably more. 

Now make the probe operate as /19 not /10, and we need 90% more capacitance than for /10.

If you increase the RIN5,6,7,8 values to 10M then the probe will be /10.9, and should compensate as you have shown

Now I need to find some 10M resistors and take the 120pF caps out of the probes...

Many thanks - always nice when weird circuits become understood

--David

drandyhaas, 2018-05-26:
This person will sell 100 of them for 1$... with free shipping... 
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F182342653535

closed 2018-05-26 by drandyhaas

drtrigon, 2018-09-11:
I'm curious: If I would buy a Haasoscope today - would there be a 1M or a 10M resistor?

drandyhaas, 2018-09-11:
By default, it will still be the (bad) 1M resistors. At least until I spin a new batch of boards. But if it's important to you, I'd be happy to sell you one with the 10M resistors instead.

drtrigon, 2019-01-07:
Can you give an estimate when the new Board v9.0 will be ready? Then I plan to order 

1x Haasoscope
1x Oscilloscope Probes
1x A Cool Screen!
1x High-speed USB Readout Board

drandyhaas, 2019-01-07:
Hi,
Good news - it's ready!
They'll be available on eBay later today. Sounds like you want a "bundle" + "usb2 board".
Best, Andy.

drtrigon, 2019-01-08:
Will that also be available from CrowdSupply? (I don't like ebay too much)

I ebay is the only option, what's the link? Is paypal available?

drandyhaas, 2019-01-08:
https://www.ebay.com/sch/i.html?_nkw=haasoscope
It will be on CrowdSupply later this month.