drandyhaas/Haasoscope #11: Document configuration interface

danielkucera, 2018-06-05:
Hi Andy,

would it be possible to document the API of the card? I mean all the options which need/can be set for the card to work. I see that there's a minimal working example in https://github.com/drandyhaas/Haasoscope/blob/master/software/serial_read.py but it's not very explanatory.

This would be useful for writing different drivers for the card (I'm thinking about writing PoC driver for PulseView) or automated data collection.

drandyhaas, 2018-06-05:
https://github.com/drandyhaas/Haasoscope/blob/master/software/commands.md

closed 2018-06-05 by drandyhaas

danielkucera, 2018-06-06:
It's quite clear, thanks. But I wonder why did you not use some predefined format, e.g.: 1B command id, 1B data length, n Bytes data (board addr, parameters), 1B checksum, or something similar.

drandyhaas, 2018-06-06:
That would have been smarter, in hindsight! The first commands I worked on didn't need extra data, so it sort of grew organically... and this is how it ended up.