drandyhaas/Haasoscope #15: Wish list

bnusbick, 2018-06-15:
Hi Dr Haas,

I think the Haasoscope is great as is, but since you keep tweaking it, I have the following wish list.  (Since it is open source, we can always add whatever we want, so it is no big deal if you choose to ignore these suggestions.)

Wishlist:
1. When you make a lot of changes, you update a brief readme file so we know if it is a bug fix or an enhancement that we should try out.

2. You implement horizontal and vertical cursors like the Syscomp CGR-201.

Also, I am thinking about downloading the new firmware to try out the logic analyzer feature, but want to wait until it is stable.  Is it working satisfactorily now, or are you planning to make changes to bring it up to snuff?

Thanks.

drandyhaas, 2018-06-15:
Glad you're enjoying it!
And thanks for the suggestions. See below..

The logic analyzer is stable AFAIK - try it out! Let me know if there's problems. It has not been heavily tested.

1. OK, I've taken a stab at such a changelog here:
https://docs.google.com/document/d/18tjrI03KipmFqbNIHOzAKYy1VVE9wWbefSHazbPP8Gc/edit?usp=sharing
and will do my best to keep it up to date. I've also linked it from the README file. Note there are separate docs (linked from that one) that track changes to the firmware, board, and android app.

2. I'm not so familiar with those cursors. What would you like to do exactly? The current canvas does show you the x,y position of your mouse, in the bottom status bar, so you can at least measure positions.

bnusbick, 2018-06-16:
Right now when I want to measure the frequency, I move the mouse to the peak of one wave, note the y value, and then move the mouse to the peak of the next wave, and subtract the first y from the second y.  What would be very useful would be to be able to click on the first peak in some manner which would place a vertical line there (cursor1), and then click on the second peak in some manner which would place another vertical line there (cursor2).  The python program would then calculate the difference in y position of the two cursors to get the interval and frequency (1 / interval).  If horizontal lines could also be placed in a similar manner, voltage differences could easily by measured by subtracting the x values of the cursors.

bnusbick, 2018-06-16:
The change log is exactly what I was hoping for. Thanks.

bnusbick, 2018-06-16:
Whoops, I meant I used the x values for measuring interval and frequency, and y values for measuring voltage.

bnusbick, 2018-06-16:
Since a right-click places a vertical dotted line and a horizontal dotted line at the point of the click, it would seem that a similar technique could be used to place just a vertical or horizontal line.

closed 2018-06-20 by bnusbick