drandyhaas/Haasoscope #16: Oversampling doesn't seem to be working correctly

bnusbick, 2018-06-15:
When I turn on oversampling, the waveform becomes more jagged instead of smoother.  It is as if the additional samples actually took place a little before the previous samples instead of after the "normal" samples.  I am not sure, but I thought this used to work better.  Maybe this occurred when you adjusted the screen display so oversampled no longer appeared as half the frequency as normal sampling.

drandyhaas, 2018-06-15:
Hmm. Are you using the latest code? git pull?
Try turning the board totally off and then on again? I've seen the board get into a confused state where it thinks it's not oversampling but the python side does...

bnusbick, 2018-06-16:
Yes, it does appear that the board is in a confused state.  When it works correctly, the frequency appears that it has been cut in half.  When it doesn't work correctly, the frequency is not cut in half.

drandyhaas, 2018-06-16:
So sometimes it works, but then later is confused? Does it always work well right after turning it on? Again, did you check you're using the latest code please by doing a git pull?

bnusbick, 2018-06-17:
Yes, I have performed github pull to get the latest code.

Today I can't get it to work correctly with oversampling at all.
I have tried exiting the program and pulling the the usb cable and rebooting the computer multiple times.

I am using a 1 MHz square wave from a signal generator:

I see the following:
It starts up correctly displaying the signal as 1 MHz.
Then I choose to trigger on channel 0.
Then I press ''O' for oversampling.  Once I do this, the frequency cuts in half and it appears jagged.  When I press 'O' again to remove oversampling, the scope still makes the frequency appear as 500 KHz, but the jaggedness is gone.  Toggling 'O' again restores the jaggedness and the frequency remains at 500 KHz. It doesn't seem to make any difference if I remove channels 1-3 from the view or not.

drandyhaas, 2018-06-17:
Strange. Maybe try a new fresh directory with the code straight from github?
Does it work if you plug your signal into channel 1 instead of channel 0, and oversample that?
Note that the timebase should automatically change from 2 us/div (or whatever it is on, as long as it's greater than 0.5us/div) to 0.5us/div when you turn on oversampling.
See video of my test here:
https://www.dropbox.com/s/aan31bykat01khn/2018-06-17%2010.38.18.mp4?dl=0

What happens if you turn on ch.2 after you turn on oversampling? Does it look like this? It should.
[image]

bnusbick, 2018-06-18:
I looked at the video. I wish I could get the same results. I created a new github clone. I do set d.serialdelaytimerwait=600, since anything less causes the app to complain. I am also running on Lubuntu 16.04.

Here is the screenshot of what I see right after beginning the program and selecting trigger on channel 1:

[image]

when I then press 'O' to get oversampling, I see the following:
[image]

When I press 'm' and expand the time base, I see the following:
[image]

When I turn off oversampling, the jaggedness decreases as shown below:
[image]

When I turn oversampling on again and enable channel 3, I see the following (two views with differing timebases):
[image]

[image]

By the way, the cheatsheet says that 'm' is available from the matplotlib toolbar.  I don't see it, is it really there?

drandyhaas, 2018-06-18:
Ahah. It looks like the scope may be working fine.
I think there is possibly high-frequency noise present in your signal being input to the scope, which you are then seeing when using oversampling, since you are sensitive to frequencies twice as large. 
Are you using a bnc coaxial cable from the signal generator straight to the scope, as in my video? Using probes, especially with the 6" ground alligator clip cable, can pick up major noise at high frequencies.
Do you have another high bandwidth scope to try instead of the Haasoscope for comparison? You may see the same high frequency noise on that scope also.

drandyhaas, 2018-06-18:
Oh, and yes, that checkmark button on the toolbar lets you change many aspects of the lines and markers.

bnusbick, 2018-06-19:
I tried using a test frequency that was generated by another FPGA board, and I do not see the jaggedness in the waveform when oversampling.  So maybe it is due to the signal generator.
Thanks for looking into it.

closed 2018-06-20 by bnusbick