drandyhaas/Haasoscope #17: Run haasocope software on OSX 10.13.5 High Sierr

salmansheikh, 2018-06-23:
I am having lots of problems getting things installed on OSX  to get the software to run.

e.g., 'OSX, Python 2.7.10

I believe sip is installed but running the HaasocopeDraw complains about it not being installed.
brew info sip
sip: stable 4.19.8 (bottled), HEAD
Tool to create Python bindings for C and C++ libraries
https://www.riverbankcomputing.com/software/sip/intro
/usr/local/Cellar/sip/4.19.8_4 (16 files, 1.1MB) *
  Poured from bottle on 2018-06-23 at 16:26:36
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/sip.rb
==> Dependencies
Recommended: python ✔, python@2 ✔
==> Options
--without-python
	Build without python support
--without-python@2
	Build without python@2 support
--HEAD
	Install HEAD version
==> Caveats
The sip-dir for Python is /usr/local/share/sip.

_**Homes-iMac:software ssheikh$ python HaasoscopeDraw.py 
Loading HaasoscopeLib.py
Traceback (most recent call last):
  File "HaasoscopeDraw.py", line 1, in <module>
    import HaasoscopeLib
  File "/Users/ssheikh/Haasoscope/software/HaasoscopeLib.py", line 18, in <module>
    import matplotlib.pyplot as plt
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/pyplot.py", line 98, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/__init__.py", line 28, in pylab_setup
    globals(),locals(),[backend_name],0)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/backend_qt4agg.py", line 13, in <module>
    from backend_qt4 import QtCore, QtGui, FigureManagerQT, FigureCanvasQT,\
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/backend_qt4.py", line 25, in <module>
    from qt4_compat import QtCore, QtGui, _getSaveFileName, __version__
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/matplotlib/backends/qt4_compat.py", line 36, in <module>
    import sip
ImportError: No module named sip**_

drandyhaas, 2018-06-25:
Are you still having issues? I'm sorry, I'm not a mac person. Hopefully someone else watching will be able to help. Googling the issue can usually lead to a solution...
This page looks pretty good... 
https://gist.github.com/patriciogonzalezvivo/77da993b14a48753efda
And talks about adding this to your path:
"export PYTHONPATH=/usr/local/lib/python:$PYTHONPATH"

salmansheikh, 2018-06-25:
Yeah but I did get the gui to start working on a windows 10 virtual machine
running in virtualbox (after installing abunch of python stuff like the
guide said) on but saw no waveforms but lost of messages in terminal that
seemed like maybe I need a USB2 device (which I ordered) or usb3 (this imac
is 2010 and doesn't support usb3 unless I get a thunderbolt to us adapter).
But even in windows VM the gui was sluggish. I will try a windows laptop. I
wonder if Linux works easy as windows...

drandyhaas, 2018-06-25:
Huh. Yes, Linux works very well.

salmansheikh, 2018-06-25:
okay, I will try on a Linux VM else install Linux natively on my iMac and
make it dual boot.

closed 2018-06-29 by drandyhaas