drandyhaas/Haasoscope #18: Can't get single event trigger to work correctly

bnusbick, 2018-06-29:
I can't get single-event trigger to trigger when it won't trigger immediately.

For example, if I configure it as follows:
  right click to set trigger
  press 'r' to remove rolling trigger
  press 'P' to toggle single mode
If I have a continuous waveform, when I press 'p', I will see a single capture of the continuous waveform, but if there is no waveform present, I don't see a single capture when the waveform occurs.  It also doesn't trigger if I press 't' to make trigger on rising edge, and then change the voltage from 0 to +3V.

Also, when I press 'p', the console almost always displays "paused False".  Every once in a while it displays "paused True".

In summary, I can't get it to trigger when there is a single event.

If single event trigger is working for you, could you show it to me in a short video like you did with my last issue?  

Thanks.

drandyhaas, 2018-06-29:
Sure, here's a video of me using it:
https://www.dropbox.com/s/7hj29qf7isy3pa3/2018-06-29%2013.53.43.mp4?dl=0

bnusbick, 2018-06-29:
Thanks, for the quick response and video, but this is not where I am having a problem.  What I would like for you to try is:

Have no signal coming in (i.e. probe grounded) and have your trigger set to rising edge of 1.5 volts.  Then move your scope probe from ground to positive 3 volts (or so) and keep it there.  Does it trigger for you?  It doesn't for me.

Thanks.

drandyhaas, 2018-06-29:
OK, please git pull from master and try now!

bnusbick, 2018-06-30:
Thanks! It works much better now.  Sometimes it doesn't catch the transition, but most of the time it does.  My testing consists of pressing 'P' to put it in get one mode, and 't' to trigger on positive edge, then momentarily touch the probe to 3 volts and seeing if it triggers.  After it triggers, I press 'p' to arm it for another round.  I will close the issue, but you may want to try the test I tried and see if it always works for you.
Thanks.

closed 2018-06-30 by bnusbick