drandyhaas/Haasoscope #22: Single Event triggering

bnusbick, 2018-10-04:
Hi Andy,
I am convinced there is a problem with single-event triggering.  My experiment consists of monitoring a switch which I set low, and then set high.  Sometimes it does not catch it going high the first time, but does on subsequent times.  I am setting it for one-event, rolling off, rising trigger, and then unpause when I want to set up for a new event.

Can you give me a clue as to where I should look to try to debug this issue?  Obviously, it is much harder to debug the hardware than software, so I would like to first somehow check to see if the events are getting to the software, but are somehow not causing a screen update.

Thanks.

drandyhaas, 2018-10-04:
If it triggers, the display will always be updated. It must not be triggering on your signal always, for some reason. What is the signal exactly? From what voltage to what voltage? Over what time interval? Does it matter what timebase you use? Where you set the trigger threshold? What about the "time over threshold" setting? (see the guide: "T followed by a number then enter: will set the number of contiguous samples above/below the trigger threshold required to fire the trigger (1 by default)" )

drandyhaas, 2018-10-04:
The samples are continuously put into a circular buffer, so the trigger is always active, except when the buffer has already triggered and is thus being read out.

bnusbick, 2018-10-05:
Hi Andy,
The attached file shows what the signal looks like.  Often times it is much dirtier, but that doesn't affect whether it catches the trigger or not.  I am monitoring it with a multimeter, and the the high voltage is 2.71 volts, and the low voltage is 0 volts.  When it doesn't trigger, the multimeter indicates that it has changed, so I am sure the voltage has changed.  It does not seem to matter what time base I use.  I have not changed the "time over threshold" setting.  I am using a clone that build that I downloaded on 10-3-2018.  The only change I made to the software was changing d.serialdelaytimerwait=600, since otherwise my computer complains about missing events.

It still fails if I change the timebase or the trigger level.  It seems to happen much less often if I set T to 2 or 5 (I was going to say never, but I did see one failure at T=2 in a couple hundred tries).  

I have included two pictures:
typical_trigger.png shows a good trigger
missed_trigger_on_up_caught_on_down.png is when missed trigger when I set switch high, but caught it when I set switch low again.

[image]

[image]

drandyhaas, 2018-10-09:
I hear you, and I'll try to reproduce this when I get a chance.

drandyhaas, 2018-10-09:
Yup, seems to work fine.

bnusbick, 2018-10-21:
Hi Andy,

I rebuilt the FPGA code with a simplified trigger as shown in the attached file.  With this file, it has never missed a high going trigger on channel 0 when in single-shot mode.  However, sometimes all the samples displayed are high, which is definitely incorrect since I always start with the signal low.  The following screenshot shows what a reasonable trigger display looks like:
[image]

The following screenshot shows all the samples high:
[image]

I think this is the reason that the submitted code has an issue, since it expects to see samples before the trigger low.

Here is the simplified oscillo.v.

[oscillo.v.txt](https://github.com/drandyhaas/Haasoscope/files/2498772/oscillo.v.txt)

drandyhaas, 2018-10-22:
Hmm. Do you have another scope you could use to debug the issue? Split your signal into the other scope and the Haasoscope, and try to get a snapshot of the signal on the other scope when it was missed by the Haasoscope. Maybe there is something pathological about the signal sometimes.

bnusbick, 2018-10-22:
Hi Dan,

I would really like to get this issue resolved.  I am quite a novice at Verilog.  Could you give me a description of the flow of getting data from the A/D chip to the PC?  From my browsing the code, I am guessing it goes somewhat as follows (I will concentrate on channel 0):

The samples come in through data_flash1 and get assigned to data_flash1_reg.
If it passes the following test:
  if (i==0) Threshold1[i] <= (data_flash1_reg>=trigthresh && data_flash1_reg<=trigthresh2);
it sets the following:
  selftrigtemp[i] = 1;
which then gets assigned to selftrig, which then gets assigned to Trigger, which then causes the following to get set:
  AcquiringAndTriggered <= 1; // Trigger? 256 more bytes and we're set
  PreOrPostAcquiring <= 1;
  wraddress_triggerpoint <= wraddress; // keep track of where the trigger happened
  state=POSTACQ;
It stays in POSTACQ until the value in samplecount is all binary 1's.  

It also seems that no matter whether it triggered or not, it is always assigning the value of data_flash1_reg to dout1.

This is where it is getting fuzzy to me.  I would guess that it should it should be storing the values from dout1 in some RAM until samplecount is all binary 1's, and then send all this data to the PC, starting with however much pretrigger samples it collected.

Since I am seeing it all high, I am guessing that the RAM is continuing to get filled after samplecount is all binary 1's, or that the pretrigger start pointer is messed up.  When I see the issue, the trigger event takes longer to complete and there is no bounce in the data, so I don't think it just a wrong starting index into the correct RAM buffer.

As you know, all the above, is in oscillo.v.  I am now starting to look at other files, but not exactly sure which are relevant.  It am guessing that I should look at WRITE1 in serialprocessor.v.

For debugging tools, I have easy access to a logic analyzer that will stream collecting 8 channels at 24 MHz for many seconds, as well as one that states that it can capture 32K 16-bit samples at 200 MHz.  I may be able to use a logic analyzer that can sample at rates much higher than 200 MHz for a couple million samples, but I would like to avoid this route.

Thanks.

drandyhaas, 2018-10-22:
Hi,
I wouldn't worry about the data flow in the firmware - it is working fine once the event is triggered. But yes, it is being stored in the FIFO buffers.
I just tried and I'm not really able to recreate your issue. I put it on the same timebase (+-256us), one-event, no rolling, and unpause for each event, and it always triggers. Things to try:
 - right click _after_ setting the time base, to set the trigger point correctly
 - then after that, do "T10<enter>" to set the number of samples (at the current timebase) that the trigger must be beyond threshold

bnusbick, 2018-10-23:
Hi Andy,
I found that it almost always correctly triggered when I added one more 0 to timercounter limit.  I then disabled the incrementing of timercounter so it will never time out, and is better still.  Maybe this is related to me having to set d.serialdelaytimerwait=600.

bnusbick, 2018-10-26:
Since disabling incrementing timercounter allows single-event ttrigger to work about 98% of the time for me, I will close this issue.

drandyhaas, 2018-10-31:
Maybe also try the latest firmware. I made some fixes to the data acquisition.

bnusbick, 2018-11-01:
Hi Andy,

This firmware works much better than before.  It catches maybe 98% of the single events.  However, a couple of times when it missed the event, and then captured my changing the switch back to low I saw the following, which is weird because I don't even have anything connected to any of the channels besides channel 0.  On a normal trace these other channels are always at 0.  Here is the weird trace after it missed catching the event:

[image]

And here is what a normal single event capture looks like:
[image]

Blake

bnusbick, 2018-11-01:
Andy,

Actually not allowing timercounter to increment performs much better for me, and it doesn't get into the state where it is displaying the wrong channels.

Blake

bnusbick, 2018-12-17:
Hi Andy,
I modified the FPGA code to bring out RxD on PIN 135 and TxD on PIN 140 to get further insight into what is happening when it fails to catch a single event.  I noticed something very interesting on the two times it failed in my limited testing.  Both times the PC started sending command 0x64 0xA while the FPGA was transmitting the acquisition data (I notice that the PC sends the 0x64 0xA command every 3 seconds while waiting for the event).

Since I am monitoring the data at the FPGA, I haven't verified that the rs232 usb chip actually sent the data to the PC, or that the PC actually got it.  I am hoping I can verify that with Wireshark.

I am also wondering if the python code clears any data that may be in the receiver buffer when it sends a new command as a recovery mechanism.  

I plan on decreasing the timeout from 3 seconds and increasing the amount of data per channel from 512 to 2048 to increase the chance of simultaneous send/receive to occur and see if missed events are always associated with this.

The only changes I made to the 12/15/18 code I cloned was modifying what came out on debug1 and debug2, and preventing timercounter from incrementing.  (I do notice that when I prevent timercounter from incrementing, the FPGA board becomes unresponsive until I reset it if I disable rolling trigger and then re-enable it again, but that I will investigate that later.)

Do you have any thoughts on what I have seen?

Thanks,

bnusbick, 2018-12-18:
Increasing the sertimeout avoids the problem.  Thanks.

closed 2018-12-18 by bnusbick