drandyhaas/Haasoscope #24: Sigrok, pulseview as desktop client

luftek, 2018-10-08:
Hello,
thank you for your work.
Do you know about https://sigrok.org/ ? It is great and in active development.

drandyhaas, 2018-10-09:
Hi. Yes, I'm aware of it. I looked a bit at what would need to be done to add support for this board, and it didn't look easy. I couldn't really even figure out where to start. I think some sigrok expert would have to do it.

luftek, 2018-10-09:
Does your board/fpga have a posibility to send data over usb?
I could try to write a device driver as there are already boards supported with uart interface.

drandyhaas, 2018-10-09:
Yes, it sends data over usb with a uart protocol.
See this file for a minimal example in python of how to setup and then get data from the board:
https://github.com/drandyhaas/Haasoscope/blob/master/software/serial_read.py
Once the board is set up, you get events by sending (over uart) two bytes: 100 10.

luftek, 2018-10-09:
Does fpga have native usb protocol? Uart is probably not fast enough for highest bitrates.

drandyhaas, 2018-10-09:
The basic uart is just 1.5 Mbps, which is ~10 Hz of events (depending on num samples per event).
But there is also a USB2 "hat" available that plugs on the board and enables higher speed uart readout at about 30 Mbps (20x faster), which gets you ~hundreds of Hz!
See the guide: https://docs.google.com/document/d/1euyzCGzQo0u9APwEeior9fpyhcYWbkNxYQaV7SuFfoE/edit?usp=sharing
and look for "USB2".

luftek, 2018-10-16:
Any chance you would lend me one of this uart oscilloscopes? Would like to try and add support in sigrok project. I already have "USB hat".

drandyhaas, 2018-10-16:
Sure! I'll put the latest firmware on it, which also has the logic analyzer functionality. 
Just send me your address.

drandyhaas, 2019-01-31:
Any progress?

luftek, 2019-02-15:
What would be the procedure to get only logic analyzer data?
0. initialization as in serial_read.py (this seems to initialize more than just logic analyzer part)
1. enable logic part ('145 5') so 5th (virtual channel) on top of all analog channels
2. receive data

drandyhaas, 2019-02-16:
I don't think there's a way to get only the logic analyzer data with the current firmware.

luftek, 2019-02-16:
Ok, I see. Not a problem.
Fooling around: https://youtu.be/BYRskn1hWNk

luftek, 2019-02-16:
I can see that drawing matplots considerably limits communication refresh rate. Probably should be done in separate Thread (https://www.tutorialspoint.com/python/python_multithreading.htm).
Now that I have smaller working version, will try to add it to sigrok

luftek, 2019-02-20:
Hi Andy, I am in the process of writing HaasoscopeLib.py alternative. I will do that before sigrok implementation, to I understand the project a bit better.
Is that ok with you?
Progress will be visible on my fork. Because I am on version v8.81 of hw, will ask you to check some things.

drandyhaas, 2019-02-20:
Ok, if you like. Just let me know when you want me to have a look.

luftek, 2019-02-20:
So first commit with a lot of constrains, and probably more.
https://github.com/luftek/Haasoscope/tree/alt

luftek, 2019-03-01:
Does setting DAC affect ADC common mode voltage setting? I don't understand the direct connection.
When I run https://github.com/luftek/Haasoscope/blob/alt/software/modules/spi.py without running initialization for DACs I don't get selected voltage on C-AIN1.
[image]

drandyhaas, 2019-03-01:
Hmm. Not sure why that would be. Dac is over i2c..

luftek, 2019-03-02:
For test I initialized ADCs and 3 out of 4 DACs. I can see common voltage setting of 0,9V on 3 capacitors but not on the 4th one. Looking at the schematic I don't see the connection. Was there any change on this parts from v8,81 to v9 ?

drandyhaas, 2019-03-02:
No change to v9.
Looks like INA1 for instance comes out of the op amp, which partly takes OFFSET1 from the dac as an input.

According to the max19506 datasheet, for the adc, INA+ and INA- are connected when vcom is being used (since it's a common voltage). 

[image]

luftek, 2019-03-10:
Learning about matplotlib.
I will need some time to chew through it. Did refactor part of ..Lib.py into smaller chunks (modules).
You did great job enabling all those features.

ifishtoo, 2019-08-13:
I see this version 9 on ebay.  How is the sigrok driver coming along?

Jeffbeck67, 2019-12-15:
Hello.
In case this gives motivation, I'm also interested into a user friendly SW :)
Any chance this get on SigRok ?

drandyhaas, 2019-12-17:
I'll have a qt based gui ready to test this week!

drandyhaas, 2019-12-18:
To try out the new Qt-based gui, do
python HaasoscopeQt.py
A lot of functionality is implemented, but it's still very rough at the moment.
Please let me know what you think!

Jeffbeck67, 2019-12-19:
Hello Andy,

I've tried on Win10 and I only have a terminal that pops up and close right after.
I installed QT but did not help.

drandyhaas, 2019-12-19:
Glad the new display is working for you.
Just doing
pip install pyqtgraph
should do it. 
As for trigger, it starts up in Auto mode. This means it refreshes every 0.2 seconds even if there is no trigger. The Normal mode disables this and waits for a trigger.

Jeffbeck67, 2019-12-21:
Hello Andy,

I have some pain making the Ext Trigger working on signle shot. In fact I managed to make it working once. Otherwise I just keep displaying the curve with a trigger that I don't see where it cames from.
I use the Qt interface for ease of use. Since it look like a scope let go beyond. My remarks:
- Test doesn't display properly on a Rasp Pi. I guess you can't do much.
- The button active/disabled have a too similar color
- The Trigger part should have a led or change color once it's fired.

drandyhaas, 2019-12-21:
If you want only an ext trigger, turn off the auto trigger by clicking the button once, enable the ext trigger, turn the trigger threshold slider down to 0.

I'll work on the other things. Thanks for your feedback.!

Jeffbeck67, 2019-12-22:
Thanks, it works!
Of cours I've tried dozen of times before :)

One thing I noticed with the Qt GUI:
- If we scroll through the channels numbers and go back the " Triggering On" and Displayed box are against ticked.

In both Qt and Python Display
- Aliasing: I sample 1 MHz signal. When the time base is too slow for that frequency, a beautiful 40 kHz and this reapear at 40 Hz.

csrl, 2019-12-23:
@luftek any chance you have or still plan to make progress on sigrok support?

Jeffbeck67, 2019-12-25:
Hello all and happy holidays,

Andy, how difficult would it be to add a file capture button to the actual Qt GUI ? In order to generate a file for Excel export ?
I guess that would be an appreciable feature.

drandyhaas, 2019-12-27:
> Thanks, it works!
> Of cours I've tried dozen of times before :)
> 
> One thing I noticed with the Qt GUI:
> 
> * If we scroll through the channels numbers and go back the " Triggering On" and Displayed box are against ticked.

Not sure I understand. The "triggering on" and "displayed" properties are specific to each channel. So if you turn them off for chan 0, for instance, and then switch to chan 1, they will be back on (checked) for chan 1. But if you go back to chan 0, they should go back off.

> 
> In both Qt and Python Display
> 
> * Aliasing: I sample 1 MHz signal. When the time base is too slow for that frequency, a beautiful 40 kHz and this reapear at 40 Hz.

Yes, that is a limitation of the hardware. There are not low pass filters, other than the true bandwidth of the device, which is fixed and large. I'm not sure what to do about that.

drandyhaas, 2019-12-27:
> Hello all and happy holidays,
> 
> Andy, how difficult would it be to add a file capture button to the actual Qt GUI ? In order to generate a file for Excel export ?
> I guess that would be an appreciable feature.

OK, now you can click "Record to file" in the File menu, and the "Stop recording" to stop. Only "displayed" fast ADC channels are recorded.
I do appreciate the feedback! Please though, make a new issue in github for each issue/request, so I can keep track, close issues, etc. Thanks.
Happy holidays!

Jeffbeck67, 2019-12-28:
> OK, now you can click "Record to file" in the File menu, and the "Stop recording" to stop. Only "displayed" fast ADC channels are recorded.

Andy, you are my absolute Santa! It works fine!
just got surprized the files are 512 Samples.
Is there a way to get the sample rate as well ? Now we cannot have a horyzontal scale on the graph.

Jeffbeck67, 2019-12-28:
> Not sure I understand. The "triggering on" and "displayed" properties are specific to each channel. So if you turn them off for chan 0, for instance, and then switch to chan 1, they will be back on [...]

I confirm on the older version it did it and not on the new one.

Concerning the update Qt GUI I can't make a proper review since I had issue on my RaspPi Linux and some displays are not as nice as before.

danielkucera, 2019-12-28:
I've tested the UI and it is very usable. But I was just thinking if it wouldn't be more progressive to have the UI in browser like this guy does here: https://www.youtube.com/watch?v=aghTg4Pggv4
Not everyone can install python with Qt but everyone definitely has a browser...

zigastegu, 2020-03-23:
Sorry @drandyhaas. I kind of got stuck in March 2019.
What has changed since?

drandyhaas, 2020-03-23:
Not much really. There is a new qt gui. And less buggy firmware. 
A sigrok interface would still be really nice!