drandyhaas/Haasoscope #25: Please make proper Use of git

S4enL0ng, 2018-10-08:
Just a random tip from a random guy which points at something.
Please tidy up the versioning as files. You have a git which does the versioning for you ;)
For example the: v9 files in the eagle subfolder. Use tags and/or branches for something like this.
Also does git let you checkout old versions of every stage without the needings to check "backup" Files or similar in this. for example in your firmware-subfolder. It will make the whole Git-Repo looking much better.
;)

drandyhaas, 2018-10-09:
I'll work on that. Sorry, I'm a git novice. (But better than Dave! :))

drandyhaas, 2018-11-26:
Some progress made on this. There's now a version9 branch, that has all the new v9 firmware and eagle stuff. The master branch has the v8 firmware and eagle stuff (but also v9 eagle stuff, since I don't want to break links people have posted to the new schematics on youtube, etc.). The software area should always be able to remain the same for master and version9 since the software should always be backwards compatible and work for any board version (using info gathered from the board to take proper actions).

closed 2020-03-14 by drandyhaas