drandyhaas/Haasoscope #27: Repeated samples?

stalinbeltran, 2018-10-16:
**What are you trying to do?**
I am saving each xydata to a file. However, when checking the data I realize there is some unexpected coincidences, so I run a python script to check for repeated data. By running this script I found the data is exactly repeated 9 to 12 times. This, of course, could be a big coincidence, but happen in a very specific way, what look suspicious. Maybe, we are reading the same buffer again and again until a real new xydata appears (I am saving the data each time it is read, inside the while loop on HaasoscopeDraw.py)

Maybe there are not new readings available, so the serial reads exactly the same previous one? (That could explain the whole problem)

**What is your OS and python version?**
I use Windows 7, Python 2.7.10

**Have you installed the prerequisites?**
Yes, and the software is running.

**To Reproduce**
Steps to reproduce the behavior:
1) Run the script included in the folowing zip (data included):
[toReproduce.zip](https://github.com/drandyhaas/Haasoscope/files/2483781/toReproduce.zip)
The data included was saved by using 
np.save(file, d.xydata)
which output is a binary python format

The Haasoscope was in default Trigger mode, triggering on one channel only.

**Expected behavior**
Each time we read the data we expect the readings vary slightly. What happens is we got the exact same set of xydata around 10 times in a sequence, what could signal the buffer is not changing at all.

**Screenshots**
Description of the following log (output of CheckResultsRepeated.py):
Column 1: actual xydata number (I gave them a sequential number). This is the xydata we are comparing to.
Column 2: number of times it is repeated in the following xydata (previous xydata are not compared)
Columns 3 (comma separated): the xydata found equal 

As you can see, the exact same xydata are always the few next ones, what is too much of a coincidence.

**OUTPUT**

********************************************************************************************
C:\Desarrollo\Haasoscope\software>python CheckResultsRepeated.py
0 3 1,2,3,
1 2 2,3,
2 1 3,
3 0
4 0
5 12 6,7,8,9,10,11,12,13,14,15,16,17,
6 11 7,8,9,10,11,12,13,14,15,16,17,
7 10 8,9,10,11,12,13,14,15,16,17,
8 9 9,10,11,12,13,14,15,16,17,
9 8 10,11,12,13,14,15,16,17,
10 7 11,12,13,14,15,16,17,
11 6 12,13,14,15,16,17,
12 5 13,14,15,16,17,
13 4 14,15,16,17,
14 3 15,16,17,
15 2 16,17,
16 1 17,
17 0
18 0
19 11 20,21,22,23,24,25,26,27,28,29,30,
20 10 21,22,23,24,25,26,27,28,29,30,
21 9 22,23,24,25,26,27,28,29,30,
22 8 23,24,25,26,27,28,29,30,
23 7 24,25,26,27,28,29,30,
24 6 25,26,27,28,29,30,
25 5 26,27,28,29,30,
26 4 27,28,29,30,
27 3 28,29,30,
28 2 29,30,
29 1 30,
30 0
31 0
32 11 33,34,35,36,37,38,39,40,41,42,43,
33 10 34,35,36,37,38,39,40,41,42,43,
34 9 35,36,37,38,39,40,41,42,43,
35 8 36,37,38,39,40,41,42,43,
36 7 37,38,39,40,41,42,43,
37 6 38,39,40,41,42,43,
38 5 39,40,41,42,43,
39 4 40,41,42,43,
40 3 41,42,43,
41 2 42,43,
42 1 43,
43 0
44 0
********************************************************************************************

**Additional context**
I was reading 256 samples only, not the default 512 (maybe that affect the readings).
HaasoscopeLib.ram_width = 8

The signal is the output of an interferometer (the signal changes all the time because of the ambient vibrations), so it is less likely it was reading a super stable signal.
Maybe the trigger is not triggering, and that makes the previous readings keep the same values.

I will be happy to help solving this problem (under your guidance of course), and maybe make a pull request to upload the solution.

Thanks for such an excellent product.

drandyhaas, 2018-10-16:
Also, what is the trigger rate from your signal (how many events do you expect to capture each second)?
And did you disable the "rolling trigger"? If the rolling trigger is active, you should get a new triggered event every 0.2s (5Hz).

stalinbeltran, 2018-10-16:
Here is the code for saving readings:
[DrawLib.zip](https://github.com/drandyhaas/Haasoscope/files/2485239/DrawLib.zip)
The np.save is into the while at HaasoscopeDraw.py
The trigger signal has 4 Hz, but refreshing frequency looks like 14 Hz in console (if I am not wrong).

I did not disable "rolling trigger" (I am sure, because I got lots of readings per second). I remember the screen updated like 1 Hz, and it looks like each update was when a different reading was available.

Today I change the saving code to avoid repeated data (so this problem does not affect my results), and it save files around once per second, with the same 4 Hz signal. Please take into account the signal frequency is not that important as it can contain spikes (it does not look like a smooth curve). The real problem is the data was exactly repeated. Those are four voltages, 3 of them changing, and it is unlikely that each sample is repeated at exactly the same time (assuming my code do the right array comparison)

drandyhaas, 2018-10-17:
Hmmm. I suspect this has something to do with the rolling trigger automatically triggering at 5 Hz, but your data has not changed, so you get "repeated events". Try disabling the rolling trigger by pressing "r". You will then only trigger when your actual trigger condition has occurred, but you will still "time out" every 3 seconds. To avoid writing out a duplicate event every 3s even if no trigger has occurred, you should check for "d.timedout" before writing out each event.

stalinbeltran, 2018-10-25:
Ok, I will. Thanks

drandyhaas, 2018-10-31:
Try the latest firmware. I made some bug fixes to the data acquisition.

stalinbeltran, 2018-10-31:
Trying to upload the new firmware, I got this error:
**C:\intelFPGA_pro\18.1\qprogrammer\bin64>quartus_pgm -m jtag -o "p;C:/Desarrollo/
Haasoscope/max10_adc_firmware/output_files/serial1_v8.sof"
Error (213013): Programming hardware cable not detected**

Obviously it is connected to both the computer and the Haasoscope. The 
USB Blaster looks correctly installed on Device Manager (win7) as "Altera USB Blaster". I am using the USB Blaster included with the Haasoscope. Altera JTAG server is running.

I tried several tricks I found on the Internet, but nothing works.

stalinbeltran, 2018-10-31:
I downloaded "Quartus Prime Programmer and Tools" only. May this be the problem?

drandyhaas, 2018-10-31:
Did you run from the command line, or use the graphical interface?

drandyhaas, 2018-10-31:
Also, the latest firmware is Haasoscope/max10_adc_firmware/output_files/serial1.sof  (not _v8)

drandyhaas, 2018-10-31:
I just added a new bit in the guide about installing the USB Blaster driver in Windows. Can you try to see if that helps please? Did you install a driver for the USB Blaster?

stalinbeltran, 2018-10-31:
Of course.

> Did you install a driver for the USB Blaster?

Yes. I used the ones available in "C:\intelFPGA_pro\18.1\qprogrammer\drivers", and let windows find the right one. I could not find then online.

Somewhere I found you should install the entire Quartus Software for it to work. Does that makes sense?

drandyhaas, 2018-10-31:
I just tried it in Windows 10 on a fresh laptop, with the Programming Tools only. In the programmer, I click "Hardware Setup..." in the top left, and select the USB Blaster. Does that not work for you?

stalinbeltran, 2018-10-31:
Looks like your bit is just what I have done. Maybe the Haasoscope has to be off while connecting the USB Blaster to the computer? I have tried the Haasoscope both ON and OFF, but can try again just to be sure.

drandyhaas, 2018-10-31:
I don't think that matters. It has to be on to program it, of course.

stalinbeltran, 2018-10-31:
> I just tried it in Windows 10 on a fresh laptop, with the Programming Tools only. In the programmer, I click "Hardware Setup..." in the top left, and select the USB Blaster. Does that not work for you?

No. Maybe is Win 7 the problem?

drandyhaas, 2018-10-31:
Possibly. You could try an older quartus version, like 15.1, and use the drivers from there. And/or try the other driver I just added to the guide/github.

stalinbeltran, 2018-10-31:
I will install Ubuntu now on my laptop, to see if it works.

stalinbeltran, 2018-11-01:
Now working in Ubuntu 16.04 LTS. This will let us try it without the Win7 bias.
The hardware was recognized as expected. However there is another error:

**root@stalin-VPCY210FL:/home/stalin/intelFPGA_pro/18.1/qprogrammer/bin# ./jtagconfig -d
1) USB-Blaster(Altera) [2-1.1]
   (JTAG Server Version 18.1.0 Build 222 09/21/2018 SJ Pro Edition)
  Unable to read device chain - JTAG chain broken

  Captured DR after reset = (1)
  Captured IR after reset = ()
  Captured Bypass after reset = ()
  Captured Bypass chain = ()
  JTAG clock speed 6 MHz**

Again, I have tried some posted tricks, without success.
Please remember I am completely new in JTAG programming, so I can be doing a basic mistake.
I have being doing this:

1)I connect the Haasoscope to the laptop (no led turns on now, in Win7 at least one led turned on)
2)I connect the USB Blaster, and now the Haasoscope turn one led on. The USB Blaster turn on led and red leds.
3)Run the command to try the connection.

I have not run the Haasoscope in Ubuntu yet (maybe It need drivers?)

Thanks

stalinbeltran, 2018-11-01:
Running quartus_pgm --auto 
I got:

root@stalin-VPCY210FL:/home/stalin/intelFPGA_pro/18.1/qprogrammer/bin# ./quartus_pgm --auto
Info: *******************************************************************
Info: Running Quartus Prime Programmer
    Info: Version 18.1.0 Build 222 09/21/2018 SJ Pro Edition
   ...
    Info: Processing started: Wed Oct 31 19:41:45 2018
Info: Command: quartus_pgm --auto
Info (213045): Using programming cable "USB-Blaster(Altera) [2-1.1]"
1) USB-Blaster(Altera) [2-1.1]
  **Unable to read device chain - JTAG chain broken**

Info: Quartus Prime Programmer was successful. 0 errors, 0 warnings
    Info: Peak virtual memory: 782 megabytes
    Info: Processing ended: Wed Oct 31 19:41:45 2018
    Info: Elapsed time: 00:00:00

drandyhaas, 2018-11-01:
The Haasoscope should be on (powered with 5V) and the power led on the scope must be on when programming it. So you need it plugged in over usb, plus plugged in over the jtag cable, ie 2 cables to the computer. And the power switch on the scope must be switched to usb.

drandyhaas, 2018-11-01:
It is possible your laptop can't deliver enough power, but if it was running OK from your laptop before, it should still be OK now. 
Yes, it's fine to use any source of 5V power, including USB battery packs. 
You can test the voltage on the scope using any 5V and GND I/O pins on the board. You can also test a 3.3V I/O pin. A multimeter will suffice.

drandyhaas, 2018-11-01:
I've also found that sometimes you need to reboot the computer on linux to get jtag working again.

stalinbeltran, 2018-11-03:
I just try to use the Haasoscope today, but it is not working anymore. Maybe I burned it down?
Is it possible I had connected the USB Blaster the wrong way? It looks foolproof, but just in case here is a [picture](https://drive.google.com/open?id=1y1yqonQSQVJIF874Ylvi3xPt9FhQlGgS) of the connector on the board.

drandyhaas, 2018-11-03:
That looks fine. Does the power led light up? Maybe you messed up the firmware?
If you're in the USA, you can ship the board to me, and I'll take a look and either fix it or send a replacement.

drandyhaas, 2018-11-03:
> The power led does not turns on when I connect the micro usb, but it turns
> on when I connect to the USB Blaster. The USB Blaster never worked, so I
> don't know if I messed up the firmware.
>
> I live abroad, so returning the Haasoscope is not an easy option. The best
> option is to make it work. Do you think it is still programmable?

It's not programmable if it can't be turned on. You know about the power
switch, right? It must be too the right to take 5V power from usb.

stalinbeltran, 2018-11-03:
I had it to the left (bnc connectors facing down). 
Now I move it to the right.
I will try to program it again.

stalinbeltran, 2018-11-04:
After many hours, finally the Haasoscope is programmed!
I tried so many things that I don't know why it worked. But I noted that the switch must be to the left (bnc connectors facing down).
After successfully programming (using a usb cell charger to power the board) I found it did not turn on when using the original usb cable. Then I replace it with a Samsung charger cable and it worked again. Maybe that explain the observed problems (now the leds turn on as expected)

How can I verify if the new firmware is installed?

drandyhaas, 2018-11-04:
Good! If you use the latest python code and firmware, it should print the firmware version as 5 during startup.

stalinbeltran, 2018-11-04:
Confirmed version 5: "got firmwareversion 5 for board 0 in 4.0 ms"
Now I can't find any repeated data. I will save hundreds of readings (as part of my work), so I will let you know how are we doing.

Thanks for your patience on this issue.

closed 2018-11-04 by drandyhaas