drandyhaas/Haasoscope #28: Slow (3 Hz) reading of a 60 Hz signal

stalinbeltran, 2018-11-04:
**What are you trying to do?**
I have a 60 Hz signal, but the reading speed is only about 3 Hz maximum, even when disabling the display of measurements (what is supposed to give the best possible reading speed)

**What is your OS and python version?**
I use Windows 7, Python 2.7.10

**Have you installed the prerequisites?**
Yes, and the software is running.

**To Reproduce**
1)Connect a 60 Hz signal to one probe of the oscilloscope. In my case, touching the probe tip produce this signal (maybe because of the noising 60 Hz environment)
2)Check for the reading frequency. I got around 3 Hz in the best case.

**Expected behavior**
Read around 10 Hz minimum (assuming it takes 100 ms to send each data array to the computer). 

**Screenshots**
[16msSignal3HzReadings](https://drive.google.com/open?id=1qW7KdIdJxXSaJfda5y_qPFRPI3MSiBJf)

**Additional context**
I am using the micro USB cable only (not USB 2 accesory board)

Thanks for such an excellent product. Keep doing it even better.

drandyhaas, 2018-11-04:
It looks like you are getting
"getdata asked for 16384 bytes and got 7904 from board 0" type of messages. See the troubleshooting section of the guide for how to work around that problem.

drandyhaas, 2018-11-07:
Were you able to solve this? Can I close the issue?

stalinbeltran, 2018-11-07:
Yes, my fault. Sorry for the inconvenience. 
I should have noted that was the amount of data what caused the low reading frequency. I was waiting to use the USB 2 adapter to test speed and show some results, but I had no time these last days.

closed 2018-11-07 by stalinbeltran