drandyhaas/Haasoscope #29: Haascope no longer recognizes by PCs

ngmacha, 2018-11-09:
I was not using my Haasoscope for several months, and now when I plug it into computer nothing even appears in Device Manager. My scope is ver.8.81 board from the Kickstarter campaign, still original firmware, and was still working OK till May. I've tried to plug it into several computers, Windows 10 as well as Windows 7, directly as well as via USB hub, changed several USB cables... Same results - nothing even appears in Device Manager!! 
Please help to identify what could be a problem - is it a faulty CH340G chip, and would it be possible to use the scope without it (connecting directly to Rx/Tx pins)? How can I check if the scope itself still working or not? It's such a good tool, what a pity if I can't use it anymore :-(

drandyhaas, 2018-11-09:
Strange. Does the power led come on?

ngmacha, 2018-11-09:
Yes, the power LED is on

drandyhaas, 2018-11-09:
Try pressing down on the ch340g chip and /or usb input when turning on the scope. It may be a loose connection?

ngmacha, 2018-11-09:
Wow, you are right, it works!! So I'll need to do some soldering... Or maybe just blowing a bit of hot gun around, what do you think?

drandyhaas, 2018-11-09:
Good. Glad we figured it out. Was it the chip or the usb port?

ngmacha, 2018-11-09:
Not sure, I pressed both :-( Will be checking in more details once I get home, my eyes are not very good without the magnifying headset...
I think I'll need to make an enclosure for it - could you advise any 3D-printed design, if you have already?

drandyhaas, 2018-11-09:
Ok. No, sorry, there's no enclosure work done so far as I know. Maybe create a new issue? And if you make progress on it share it there, thanks!

closed 2018-11-09 by drandyhaas

ngmacha, 2018-11-09:
OK, I'll try to design something and post here. Thank you again and wish you a nice weekend.

luftek, 2018-11-09:
Printable 2 part enclosure would be nice. Just though about it today. :) +1 for @ngmacha

@rdandyhaas but else, its best to design next revision of pcb around enclosure :)

ngmacha, 2018-11-09:
So the culprit is the CH340G chip, indeed - it's seated a little skewed and hence not properly soldered - you can see in the photo that pin 16 is barely touches the PCB trace:
[image]
A bit of heat-gun will fix this, I think...