drandyhaas/Haasoscope #45: Can't scan JTAG chain. Error code 87

polachz, 2019-03-19:
Hallo.

I have tried to update firmware into my Haasoscope board (V 8.81) I use WIndows version and the Blaster driver provided by you to avoid the BSOD.
Then I have got this error on the command line. What can be wrong, please?

drandyhaas, 2019-03-20:
You also need to power the board through the normal usb connection (or other 5V source). The power through the jtag cable is not sufficient.

polachz, 2019-03-20:
Ok, thank you for the explanation. I'll try firmware update again today afternoon. Maybe you can improve the guide by this piece of information? I Thought about additional power, but was not sure if it si necessary ot not and if it/s not risky.

drandyhaas, 2019-03-20:
OK, clarified the guide. Thanks!

closed 2019-03-20 by drandyhaas

polachz, 2019-03-21:
I can confirm that additional power source solves the problem. Thank you for quick support!