drandyhaas/Haasoscope #46: Haasoscope not communicating

anj1, 2019-04-03:
When I try to open up HaasoscopeDraw.py, I get this error:

```
Loading HaasoscopeLib.py
matplotlib backend is TkAgg
Loading HaasoscopeLib.py
matplotlib backend is TkAgg
num main ADC and max10adc bytes for all boards =  2048 and 0
rate theoretically 10.89 Hz over serial
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=2-1
connected serial to /dev/ttyUSB0 , timeout 3.0 seconds
got firmwareversion 0 for board 0 in 100.3 ms
```
After this all that happens is an empty graphics window pops up and nothing displays.

And the minimal serial reading script doesn't work either.

Very occasionally (e.g. once every ten times) it does work. But most of the time it gives the above error.

I'm on Linux (Ubuntu) using python 2.7.

drandyhaas, 2019-04-03:
Do you have a v8 or v9 board?
Could it be the usb cable? Or usb port on the computer?

anj1, 2019-04-03:
I've got the v9.01 board. I've tried with different USB cables and computers. All give the same issue.

anj1, 2019-04-03:
Also when I try the serial_read_versions.py script (after changing the serial port to '/dev/ttyUSB0'), I get:

```
got uniqueID  for board 0  in 1001.28 ms
got firmwareversion 0 for board 0 in 100.73 ms
```
Again, rarely it does give the correct board no, (which is 8098227110eec888), but most of the time it just gives 0 or a wrong number, like 8098227110fe7c1e.

drandyhaas, 2019-04-03:
Rarely the 50MHz oscillator can not be soldered well. Do you have a hot air soldering station? Just heat up the oscillator (right to the left of the fpga) and push it down a little onto the board better.

anj1, 2019-04-04:
I don't have a heat gun, is it possible to do the repair with a very fine tip iron?

drandyhaas, 2019-04-04:
I think it is possible. The best might be to use a little solder on the top of the device and heat up the  whole thing.

anj1, 2019-04-04:
Ok I tried that and it seems to have fixed the problem.

anj1, 2019-04-05:
After sitting for a day unused, the problem has appeared again, so I am reopening the issue.

I am not convinced that the problem is entirely due to the oscillator, as I can see the 50 MHz signal on the FPGA's input using another oscilloscope.

Furthermore, what often happens is that a few seconds after starting the GUI (with everything being normal), suddenly it reports 0 data coming from the board, and the GUI freezes.

drandyhaas, 2019-04-08:
Maybe it is still not soldered very well. If it loses clock for just a split second, it could cause some strange issue on the board.

anj1, 2019-04-09:
I don't see any reason to think it might be the oscillator; again I can see the signal very clearly and cleanly and I can't see any noise/breaks/glitches in the oscillator output. And during soldering it clearly 'slunk' itself into place indicating proper reflow.

I think it may be related to the software or firmware, as it seems to happen more often when other USB devices are plugged in. And happens rarely when there are no other USB devices. Any ideas?

drandyhaas, 2019-04-09:
I've never heard of there being such issues with software or firmware. When it works, do you see firmware version 16 reported?
Where are you located? I'm happy to send a replacement if you send back yours.

anj1, 2019-04-19:
Firmware version is 16. I'm in New Zealand. Yes I would like a replacement please.

drandyhaas, 2019-04-19:
Did you buy it on eBay? You can open a return/replacement there. 
Otherwise, please try to arrange with CrowdSupply. Please let me know if you have any problems.

closed 2019-12-06 by drandyhaas