drandyhaas/Haasoscope #47: Glitchy signal showing 'spikes'

anj1, 2019-04-05:
I always seem to get 'glitches' or 'spikes' when looking at a signal. This is an example of looking at a 1 kHz sine wave from a signal generator. Looking at it with another oscilloscope, the signal is fine. I've attached pictures to show what I mean:

[1 kHz sine wave](https://imgur.com/IEqemzP)

[Same signal, zoomed in more](https://imgur.com/Vr5W4GP)

[Same signal, zoomed in much more, showing the glitch in detal](https://imgur.com/P8J0Pvz)

The glitches seem to happen during zero-crossings, irrespective of the signal frequency. For a 10 kHz signal, for example, I also get glitches around the zero crossings.

I also get the glitches in both DC and AC coupling mode.

anj1, 2019-04-05:
Additionally, I only get the glitches on channel 0, not on any of the other channels.

drandyhaas, 2019-04-08:
Strange. Can you try changing the DC offset of the other channels (move them up or down) and see if that affects the spikes on channel 0?

closed 2020-03-14 by drandyhaas