drandyhaas/Haasoscope #51: Haasoscope seems wiped after trying the FT232H USB extension board

ripzorn, 2019-08-07:
Hi 

I recently bought the extension board for the Haasoscope (I have version 9.01) on ebay (https://www.ebay.com/itm/282971875030) - It didn't include the USB2 extension cable, so I used another one - after plugging it in my Haasoscope I tried running the python script but got no response from the board. The I reattached the OLED display. But not the board seems to be wiped. I don't get a picture on the OLED display anymore and when I execute the python scripts I get no response back. The only LED that has light is the power one. The device is recognized by windows and is shown in the device manager. 
I'm using windows 10, Quartus prime version 18.1.0 and python 3.7. I had everything up and running before I installed the USB extension board. 
I have tried flashing the haasoscope with the provided pof files, but that doesn't seem to changes anything.
 
Any suggestions? 

Currently, I trying to compile the latest Max10adc firmware using my version of Quartus, but I get an error : 
" Error (18496): The Output outclk_pin in pin location 28 (pad_9453) is too close to PLL clock input pin (clock_ext_osc) in pin location 27 (pad_0)"

drandyhaas, 2019-08-07:
You tried a flashing with serial1_scope_9.0_v16_backup, and it succeeded?

ripzorn, 2019-08-07:
Just tried it and it passed. 
But still blank OLED and out response in scripts.

drandyhaas, 2019-08-07:
The compilation error you get with quartus comes from using v18.1 to compile the v8 firmware. Make sure you use the version9 branch of github for the v9 board.

drandyhaas, 2019-08-07:
It's possible some hardware issue occurred with your board. Where are you located? If you ship it back to New York, I can send you a replacement.

ripzorn, 2019-08-08:
Arh okay. The Version9 is compiling, but the end result is the same.

I'm located in Denmark (Europe) but shipping should be possible. Do you have a shipping formula or where do I look for returning the board?

drandyhaas, 2019-08-12:
You can ship to :
Andrew Haas
726 Broadway
Room 1005
New York, NY
10003

ripzorn, 2019-09-30:
> You can ship to :
> Andrew Haas
> 726 Broadway
> Room 1005
> New York, NY
> 10003
Finally got the device ship to you. It should be there within some days expect.

drandyhaas, 2019-10-14:
No package yet. Do you have a tracking number?

ripzorn, 2019-10-14:
It's a nordic postal service Postnord, and the trance and trace number is RR186092667DK.
According to them, the package was delivered on the 9th of October.

drandyhaas, 2019-10-15:
Hmmm. I don't see it at my office yet. 
It's possible it is on its way from the central University mail center to here still. It's been a 3-day weekend here in the US. 
If I don't see it today or tomorrow, I'm not sure what to do.

drandyhaas, 2019-10-15:
I just showed up!

ripzorn, 2019-10-15:
Oh Good, you had me worried about getting lost in shipping. 

I also saw that it was working the first time in plugged it in after a while (my best guess is that a capacitors needs to discharge for a day or so). But I had it running connected to the PC with the python framework running for a couple of minutes before it froze and then I couldn't get it functioning correctly - as described above.

drandyhaas, 2019-10-16:
It's been running fine all night, with the mini-display too. I don't see any issue!

ripzorn, 2019-10-16:
Well that's anoying. 
The last I remember was that it often happened when I was playing around with the time scales and voltage levels in the python frameworks. E.g., zooming out to max for a while and going back and forth. 
And that I was only using one probe on the left channel.

ripzorn, 2019-10-21:
Strange

drandyhaas, 2019-10-21:
I can send you a replacement board...
Just do it on ebay so we can use the global shipping program. I'll just lower the price to $1 or something for you.

ripzorn, 2019-10-28:
Okay, I think I would like to do that then. How do I do it via ebay then?

drandyhaas, 2019-10-30:
Send me a message about the Haasoscope in the ebay app or website. I can then send you a special offer via ebay.

ripzorn, 2019-10-31:
Hey Andrew thanks for the help and offer, however, I was under the impression that you would be sending/covering for the return shipment. If this is not the case, then I would rather make a RMA request, since the total product cost will be nearly doubled.

drandyhaas, 2019-10-31:
How much is the return shipping through ebay? Yes, I will pay for the return shipping. I can reimburse you via PayPal or something.

ripzorn, 2019-11-01:
Hey Andrew, that would be much appreciated. 
On the offer you made, I thought I saw $23 USD, but just looking at a new order now, it says US $20.81, so my guess is somewhere between that.

drandyhaas, 2019-11-02:
Ok, so make the purchase and then I will reimburse you.

ripzorn, 2019-11-29:
Should I just paste my paypal or how do you prefer?

drandyhaas, 2019-11-30:
Email me your shipping receipt and password.

logictechs, 2020-06-04:
I believe I'm having the dreaded CH340 USB to serial problems.  The device shows up fine in Windows 10 device manager.  I can run the serial_read_versions.py ok a couple times before I start getting timeout issues.  Seems to happen on my Raspberry Pi 4 also.  I also wrote a Visual Studio VB program for testing.  The same happens.  To solve the timeout issue I simply unplug the board from the USB port and return it after a couple seconds.  Can someone confirm this board really works as expected by using the USB2 interface or Bluetooth?  I was thinking of using an ESP8266 or ESP32 for TCP to serial connection.  I cannot get the HaasoscopeDraw.py to display on 2 Windows 10 machines.  On my Raspberry Pi 4 it displays but doesn't seem to be getting data correctly.  Again back to the CH340 issue.  I'm wondering if you'd consider using a FTDI chip instead of the CH340?  As long as it's genuine, it would seem to perform better.

drandyhaas, 2020-06-04:
Do you have a v8 or v9 board?
The common failures are the soldering of either the ch340 on v8 or the oscillator on v9.
You can try pushing down on the ch340 when it's failing to see if that helps, and if so you may need to touch up the soldering. Easy. 
You can tell its the oscillator because the clock signal from pin x of the oscillator will disappear or go wonky. The oscillator is a little harder to fix. If you have a hot air soldering station it's easy. Just heat it up, press down, and cool.

logictechs, 2020-06-04:
Hi Andy,

Thanks for the response and info.  I received the V9.01 kit about 2 weeks
ago now from Mouser.  I've reheated the soldering the 12Mhz crystal and the
caps near the CH340G chip.  I'll attempt reheating the osc at the Max 10
chip.  The pins of the CH340G chip seem solder well but I'll reheat them
also.

Regards,

logictechs, 2020-06-04:
After reheating the osc at the Max 10 communication seems very reliable now.  Still can't get haasoscopeDraw.py to work.  The serial_ python programs all work consistently now at least.  Thanks for the tip Andy!

drandyhaas, 2020-06-04:
Good. What's the problem with HaasoscopeDraw.py?

logictechs, 2020-06-04:
> There's also HaasoscopeQt.py now, which is better for most things, BTW.

I'll try that again when I get back to working with the Pi 4.  I'm now
fighting with my main Windows 10 machine about the CH340G driver.  It shows
up as a port but won't allow it to open.  On my bench PC it still works
ok.
Regards,

closed 2021-04-17 by drandyhaas