drandyhaas/Haasoscope #53: trigger threshold and voltage offset

bnusbick, 2019-10-14:
Hi Andy,
I am converting your haasoscope to use pyqtgraph as a GUI instead of matplotlib.  I have trigger working, but am confused by some behavior I see.  If I set the voltage offset with the calibration values for my board, it triggers as I expect it to.  However, if add a negative voltage offset such as -0.5, it still triggers at the same voltage.  I it would require a voltage 0.5 volts higher. Is the trigger comparison made before the voltage offset is applied?

In case you are interested, I have attached a screenshot of my WIP pyqtgraph GUI.

Regards,
Blake
[image]

drandyhaas, 2019-10-14:
Very nice!
Hmmm. How exactly are you applying the voltage offset?

bnusbick, 2019-10-14:
Hi Andy,

I see that the lowDacLevel DC calibration value that causes the ADC reading to be 128 when I short the probe to ground is 2101.  By experiments, I see that if I add -345 per volt offset, the ADC value changes by 255 / 7 when the probe tip is still shorted to ground.  For example,  If I set my offset to -1 volt, I add 345 to 2101, and the ADC value I read is about 165.  When I say write, I call a function that is basically a copy of your setdac function in HaasoscopeLib.py with my value, which would be 2101 for no added offset, and 2446 for an offset of -1 volt.

Regards,
Blake

ngmacha, 2019-10-14:
What a nice GUI! Would it be available, can you share it, @bnusbick ?! I am not a great Python coder, but if needed I'm ready to help, at least to test on my Haasoscope!..

bnusbick, 2019-10-14:
I wouldn't mind sharing it once I get it stable.  Currently I see strange behavior way too often.  Also, even though this has nothing to do with what I do at work and I am using no company resources, I need to make sure I would not somehow be violating company rules.

ngmacha, 2019-10-16:
Please let me know when you'll be able/ready to share it! Thanks in advance - Mac Ha
> I wouldn't mind sharing it once I get it stable. Currently I see strange behavior way too often. Also, even though this has nothing to do with what I do at work and I am using no company resources, I need to make sure I would not somehow be violating company rules.

bnusbick, 2019-10-16:
Hi Andy,

Can you confirm that the offset is applied before the ADC?  If so, since this haasoscope has a range of -3.5 to 3.5 volts, is a safe way to measure signals that range from 0 to 5 volts to set the offset to -3 volts before applying the voltage that ranges from 0 to 5 volts?

As for the strange triggering when I set the offset, experiments with my pyqtgraph setup seem to show that strange behavior occurs when I send commands after sending a command to acquire data (command 100 followed by command 10) which gets no response.  For example, if I switch to and from rolling trigger when it is not receiving data because it doesn't meet the trigger, it doesn't seem to be switching.  I know it is not switching because in rolling trigger mode it should get the data whether it meets the trigger or not.  I also noticed that the trigger offset sometimes not being applied  when rolling trigger is false.

I can mostly (but not always) get around this issue by waiting (I have tried ranges from 1.6 to 3.0 seconds) the first time I send a UART or I2C command after a get data command didn't return any data.

When I looked at your verilog code many months ago, I seem to recall you had some timeout value.  I am wondering if this has something to do with it.

I am going to go as far as I can treating the FPGA  as a black box, but I think I am quickly approaching that point.  I think to understand what is really going on, I am going to need to get visibility into what the FPGA code is seeing and doing.  Many months ago I fumbled around enough to use your two FPGA diag pins as a mirror of the UART RX and TX that I could send to a logic analyzer.  I think I need to do something similar, and maybe also re-purpose your logic analyzer outputs too.  However, I had to reformat my computer since then and lost all my changes, so I am going to have to start over again from scratch.

Regards,
Blake

drandyhaas, 2019-10-16:
> Can you confirm that the offset is applied before the ADC? 

Yes, the offset is produced by the DAC, and mixed with the signal in the op-amp, then fed to the ADC.

> If so, since this haasoscope has a range of -3.5 to 3.5 volts, is a safe way to measure signals that range from 0 to 5 volts to set the offset to -3 volts before applying the voltage that ranges from 0 to 5 volts?

Yes.

> As for the strange triggering when I set the offset, experiments with my pyqtgraph setup seem to show that strange behavior occurs when I send commands after sending a command to acquire data [...]

Maybe just keep it in rolling trigger mode when sending commands.

> 
> When I looked at your verilog code many months ago, I seem to recall you had some timeout value. I am wondering if this has something to do with it.

Perhaps.
What board version (8 or 9) and firmware version are you using?

> I am going to go as far as I can treating the FPGA as a black box, but I think I am quickly approaching that point. I think to understand what is really going on, I am going to need to get [...]

OK, good luck. The firmware is really not too hard to work with once you get used to it. Start a new issue ticket if you need help with it please.

bnusbick, 2019-10-18:
ngmacha,

I will send you the code if you send me an email address.  I don't see a way to privately contact you by your github id.

ngmacha, 2019-10-18:
@bnusbick You may send to ngmacha (at) gmail (dot) com 
Thanks a lot in advance!
> ngmacha,
> 
> I will send you the code if you send me an email address. I don't see a way to privately contact you by your github id.

closed 2019-10-19 by bnusbick