drandyhaas/Haasoscope #55: error running software

csrl, 2019-12-22:
I have hardware v9.01.  Is there a known working version of the software I can use?

```
[Haasoscope]$ python2 --version
Python 2.7.17
[Haasoscope]$ $ git show --summary
commit 2aa6fdd2cb4648d8fe72a57867999ac4fca78c22 (HEAD -> master, origin/master, origin/HEAD)
Author: Andy Haas <drandyhaas@gmail.com>
Date:   Fri Dec 20 13:19:15 2019 -0500

    trigger points

[Haasoscope]$ cd software/
[software]$ python2 HaasoscopeDraw.py 
Loading HaasoscopeLib.py
('Linux', 'ss9', '4.19.81-1-lts', '#1 SMP @1572357778', 'x86_64')
matplotlib backend is agg
Loading HaasoscopeLib.py
('Linux', 'ss9', '4.19.81-1-lts', '#1 SMP @1572357778', 'x86_64')
matplotlib backend is agg
num main ADC and max10adc bytes for all boards =  2048 and 0
rate theoretically 35.95 Hz over serial
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=1-1.2
connected serial to /dev/ttyUSB0 , timeout 3.0 seconds
got firmwareversion 15 for board 0 in 2.42 ms
minimum firmwareversion of all boards is 15
rolling trigger
num samples is 512
send increment is 0
do highres is True
trigger time over/under thresh now 1 and usedownsamplefortriggertot is True
serialdelaytimerwait is 100
tell SPI setup: 131  8 0
tell SPI setup: 131  6 16
tell SPI setup: 131  4 36
tell SPI setup: 131  5 36
tell SPI setup: 131  1 0
usb2 connection not available
got uniqueID 80982271106e805c for board 0 , len is now 1
current directory is /home/user/Projects/Haasoscope/software
reading calibrations for board 0 , channels 0 - 4
read calib/calib_80982271106e805c.json.txt
calib was written using firmware version 15
do highres is False
shut down adcs
bye bye!
Traceback (most recent call last):
  File "HaasoscopeDraw.py", line 43, in <module>
    if not d.getchannels(): break
  File "/home/user/Projects/Haasoscope/software/HaasoscopeLib.py", line 1837, in getchannels
    self.on_running(self.ydata, bn) #update data in main window
  File "/home/user/Projects/Haasoscope/software/HaasoscopeLib.py", line 1251, in on_running
    self.lines[thechan].set_xdata(xdatanew)
  File "/usr/lib/python2.7/site-packages/matplotlib/lines.py", line 1213, in set_xdata
    self.stale = True
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 241, in stale
    self.stale_callback(self, val)
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 68, in _stale_axes_callback
    self.axes.stale = val
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 241, in stale
    self.stale_callback(self, val)
  File "/usr/lib/python2.7/site-packages/matplotlib/figure.py", line 60, in _stale_figure_callback
    self.figure.stale = val
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 241, in stale
    self.stale_callback(self, val)
  File "/usr/lib/python2.7/site-packages/matplotlib/pyplot.py", line 577, in _auto_draw_if_interactive
    fig.canvas.draw_idle()
  File "/usr/lib/python2.7/site-packages/matplotlib/backend_bases.py", line 2055, in draw_idle
    self.draw(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/matplotlib/backends/backend_agg.py", line 437, in draw
    self.figure.draw(self.renderer)
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/matplotlib/figure.py", line 1493, in draw
    renderer, self, artists, self.suppressComposite)
  File "/usr/lib/python2.7/site-packages/matplotlib/image.py", line 141, in _draw_list_compositing_images
    a.draw(renderer)
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/matplotlib/axes/_base.py", line 2635, in draw
    mimage._draw_list_compositing_images(renderer, self, artists)
  File "/usr/lib/python2.7/site-packages/matplotlib/image.py", line 141, in _draw_list_compositing_images
    a.draw(renderer)
  File "/usr/lib/python2.7/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/matplotlib/lines.py", line 738, in draw
    self.recache()
  File "/usr/lib/python2.7/site-packages/matplotlib/lines.py", line 661, in recache
    self._xy = np.column_stack(np.broadcast_arrays(x, y)).astype(float)
  File "/usr/lib/python2.7/site-packages/numpy/lib/stride_tricks.py", line 259, in broadcast_arrays
    shape = _broadcast_shape(*args)
  File "/usr/lib/python2.7/site-packages/numpy/lib/stride_tricks.py", line 193, in _broadcast_shape
    b = np.broadcast(*args[:32])
ValueError: shape mismatch: objects cannot be broadcast to a single shape
```
Or when trying python 3 it is a little better - the UI screen flashes momentarily before crashing:

```
[Haasoscope]$ python --version
Python 3.8.0
[Haasoscope]$ git checkout python35 
Switched to branch 'python35'
[Haasoscope]$ cd software/
[software]$ python HaasoscopeDraw.py 
Loading HaasoscopeLib.py
posix.uname_result(sysname='Linux', nodename='ss9', release='4.19.81-1-lts', version='#1 SMP @1572357778', machine='x86_64')
('matplotlib backend is', 'Qt5Agg')
HaasoscopeDraw.py:2: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Loading HaasoscopeLib.py
posix.uname_result(sysname='Linux', nodename='ss9', release='4.19.81-1-lts', version='#1 SMP @1572357778', machine='x86_64')
('matplotlib backend is', 'Qt5Agg')
('num main ADC and max10adc bytes for all boards = ', 2048, 'and', 0)
('rate theoretically', 35.95, 'Hz over serial')
('/dev/ttyUSB0', ':', 'USB2.0-Serial', ':', 'USB VID:PID=1A86:7523 LOCATION=1-1.2')
('connected serial to', '/dev/ttyUSB0', ', timeout', 3.0, 'seconds')
('got firmwareversion', 15, 'for board', 0, 'in', 2.26, 'ms')
('minimum firmwareversion of all boards is', 15)
rolling trigger
('num samples is', 512)
('123 send increment is', 0)
('143 do highres is', True)
('129 trigger time over/under thresh now', 1, 'and usedownsamplefortriggertot is', True)
('serialdelaytimerwait is', 100)
('tell SPI setup: 131 ', 8, 0)
('tell SPI setup: 131 ', 6, 16)
('tell SPI setup: 131 ', 4, 36)
('tell SPI setup: 131 ', 5, 36)
('tell SPI setup: 131 ', 1, 0)
usb2 connection not available
('got uniqueID', '80982271106e805c', 'for board', 0, ', len is now', 1)
('current directory is', '/home/user/Projects/Haasoscope/software')
('reading calibrations for board', 0, ', channels', 0, '-', 4)
('read', 'calib/calib_80982271106e805c.json.txt')
('calib was written using firmware version', 15)
('theboard:', 0, ' num_board:', 1, ' chan:', 0, ' num_chan_per_board:', 4)
('theboard:', 0, ' num_board:', 1, ' chan:', 1, ' num_chan_per_board:', 4)
('theboard:', 0, ' num_board:', 1, ' chan:', 2, ' num_chan_per_board:', 4)
('theboard:', 0, ' num_board:', 1, ' chan:', 3, ' num_chan_per_board:', 4)
('143 do highres is', False)
shut down adcs
bye bye!
Traceback (most recent call last):
  File "HaasoscopeDraw.py", line 44, in <module>
    if not d.getchannels(): break
  File "/home/user/Projects/Haasoscope/software/HaasoscopeLib.py", line 1901, in getchannels
    self.drawtext() #redraws the measurements
  File "/home/user/Projects/Haasoscope/software/HaasoscopeLib.py", line 747, in drawtext
    self.texts[0]=(self.ax.text(xpos, height, self.chantext(),horizontalalignment='left', verticalalignment='top',transform=self.ax.transAxes))
  File "/home/user/Projects/Haasoscope/software/HaasoscopeLib.py", line 717, in chantext
    if self.testBit(self.switchpos[theboard],chanonboard):
TypeError: list indices must be integers or slices, not float
```

drandyhaas, 2019-12-23:
I'm not aware of any issues with the head. You can try the new qt gui though.
python HaasoscopeQt.py
Might need
pip install pyqtgraph
first.

csrl, 2019-12-23:
A screen loads with HaasoscopeQt.py.  I'll try using that.

For HaasoscopeDraw, perhaps there is a bug/incompatibility in the dependency packages.  Can you provide the dependency package versions you are using.  

It is a bit concerning that this new UI is being implemented in python 2 instead of python 3.  Unless a packaged installer will be created, this new UI will not be usable for long, as soon distributions will quit shipping python 2.  Please consider moving to python 3.

csrl, 2019-12-23:
HaasoscopeQt.py crashes anytime the mouse wheel/scroll occurs.

```
Traceback (most recent call last):
  File "HaasoscopeQt.py", line 219, in wheelEvent
    if event.delta()>0: self.uppos()
AttributeError: 'QWheelEvent' object has no attribute 'delta'
Aborted (core dumped)
```

drandyhaas, 2019-12-23:
Thanks. Easy for you to fix. What operating system?

csrl, 2019-12-23:
Archlinux, `Linux ss9 4.19.81-1-lts #1 SMP @1572357778 x86_64 GNU/Linux`

drandyhaas, 2019-12-27:
Maybe you picked up the PyQt5 library somehow? Anyways, I added some protection for this in the head. Please try.

csrl, 2019-12-27:
Well, yes.  Qt4 has been unsupported since 2015.  Qt5 would be the default assumption at this time.

Now that you mention it, I search the Haasoscope doc, and I do find a reference to qt4 under setup on Mac section.

Could you update the doc with all dependency versions in use?  It seems a lot of outdated software is in use, so having specific versions listed will allow some possibility of success in getting it working on more recent systems.

csrl, 2019-12-28:
[wheelscroll_patch.txt](https://github.com/drandyhaas/Haasoscope/files/4006710/wheelscroll_patch.txt)
The attached patch fixes wheel scrolling for qt5.

closed 2021-07-05 by drandyhaas