drandyhaas/Haasoscope #58: ADC Accuracy

Jeffbeck67, 2020-01-18:
Hello,
I've been measuring the precision of the ADCs that for 8 bits was supposed to be perfect and it seems not.
Protocol: 250 kHz triangle, 8,2Vpkpk to have full swing.  Recording 512 Samples and cropping the data to have 1 output code increase per sample.
[image]

I get up to 4 bits error. Is that possible ? 
[image]

I'm not familiar with ADC testing and my terminology is likely wrong.

drandyhaas, 2020-01-18:
First of all, that doesn't look terrible.
But how many bits is your function generator? 
Also, you could try smaller voltage swing, or slower sleep, to see if that changes things.

Jeffbeck67, 2020-01-19:
Usual question of test devices vs DUT. Frankly I don't know. It's given for 12 bit.
Below a 40 us ramp that is RC filtered. significantly better. 
that was a good lesson, I did not expect a 8 bit ADC to be so tricky to test.
[image]

closed 2020-01-19 by Jeffbeck67