drandyhaas/Haasoscope #59: dp_ram behavior

bnusbick, 2020-03-01:
Hi,

I noticed the following line in your verilog code:
     if (nsamp>4095) nsamp=4095; //max of 4096 samples stored in the ram (12 bit writesamp address), and one less is needed (not sure why...)

As I said earlier, I am attempting to rewrite your haasoscope so I can assign up to 32KB to one channel and use one top-level file that includes all the others as components so I don't need a .bdf file.

As part of my effort to understand all the components, I performed some experiments on lp_ram.  I notice that I must wait two clock cycles for the data to show up after I set the address on read.  I have a suspicion this is what is causing the issue in your verilog code.  Below is my test code, and I have attached a screen shot from my logic analyzer.  The address is updated in state 4, but the data doesn't change until state 6. (Using your logic analyzer input pins as debug GPIOs plus debug1 and debug 2 gives me 10 debug GPIOs.)

Here is my test code snippet (I know it is horribly inefficient, which stems from my being a novice, lots of experiments, and not being quite sure how dp_ram works):

	always @(posedge clk) begin
		debug[2:0] <= rdaddress_sig[2:0];
		debug[3:3] <= q_sig[0];
		debug4 <= q_sig[1];
		debug5 <= q_sig[2];
		{debug9,debug8,debug7,debug6} <= test_ram_state; 
		
		case (test_ram_state)
			0: begin
				rdaddress_sig <= 0;
				wraddress_sig <= 0;
				wren_sig <= 0;
				rden_sig <= 0;
				test_ram_state <= 1;
				data_sig <= 2;
			end
			1: begin
				wren_sig <= 1;
				test_ram_state <= 2;
			end
			2: begin
				wraddress_sig <= wraddress_sig + 1;
				data_sig <= data_sig+ 1;
				if (wraddress_sig == 7)
					test_ram_state <= 3;
			end
			3: begin
				wren_sig <= 0;
				rden_sig <= 1;
				test_ram_state <= 4;
			end
			4: begin
				if (rdaddress_sig == 6)
					test_ram_state <= 7;
				else begin
					test_ram_state <= 5;
				end
			end
			5:
				test_ram_state <= 6;
			6: begin
				rdaddress_sig <= rdaddress_sig + 1;
				test_ram_state <= 4;
			end
			7:
				test_ram_state <= 7;
		endcase
	end

[image]

drandyhaas, 2020-03-03:
Ok, possibly. Anyways, nice to see you're getting quite good with verilog
and quartus.

bnusbick, 2020-03-03:
if I don't register the read output port, the data changes in state 5.

closed 2020-03-03 by bnusbick