drandyhaas/Haasoscope #60: Worked once, and then never responds again

merckhung, 2020-09-02:
**What are you trying to do?**
I just received the board today. I see the board worked (connected ch0 only) twice or so (after reconnecting USB cable for a several times), and then on a occasion I saw the QT frond-end plots lots of noises of all 4 channels. After that, I never get the board to work again. Is there a way to troubleshoot this problem? Thank you.

**What is your OS and python version?**
Ubuntu Linux

Loading HaasoscopeLib.py
('Linux', 'lnx', '5.4.0-45-generic', '#49-Ubuntu SMP Wed Aug 26 13:38:52 UTC 2020', 'x86_64')
matplotlib backend is TkAgg
Loading HaasoscopeLib.py
('Linux', 'lnx', '5.4.0-45-generic', '#49-Ubuntu SMP Wed Aug 26 13:38:52 UTC 2020', 'x86_64')
matplotlib backend is TkAgg
num main ADC and max10adc bytes for all boards =  2048 and 0
rate theoretically 35.95 Hz over serial
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=1-4
connected serial to /dev/ttyUSB0 , timeout 3.0 seconds
got firmwareversion 0 for board 0 in 100.4 ms
minimum firmwareversion of all boards is 0
rolling trigger
num samples is 512
send increment is 0
do highres is True
trigger time over/under thresh now 1 and usedownsamplefortriggertot is True
serialdelaytimerwait is 100
tell SPI setup: 131  8 0
tell SPI setup: 131  6 16
tell SPI setup: 131  4 36
tell SPI setup: 131  5 36
tell SPI setup: 131  1 0
usb2 connection not available
getID asked for 8 bytes and got 0 from board 0
current directory is /home/merck/Code/Haasoscope/software
failed to get board ID for board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
[... 138 lines cut ...]
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 0 from board 0
don't have channel 0 on board 0
serial com failed!
failed to talk to board when cleaning up!
bye bye!

merckhung, 2020-09-02:
[image]

drandyhaas, 2020-09-02:
Sorry for the trouble.
A few others have had issues with the soldering of the 50 MHz oscillator on v9 of the board. It's the little square metal package just at the bottom left of the fpga. If you have a hot air soldering tool, try reflowing it. Or use a normal solder iron and heat it up and try to reflow it. Can you try?

merckhung, 2020-09-02:
Appreciate your helps. It works with no problem now.

Loading HaasoscopeLib.py
('Linux', 'lnx', '5.4.0-45-generic', '#49-Ubuntu SMP Wed Aug 26 13:38:52 UTC 2020', 'x86_64')
('matplotlib backend is', 'TkAgg')
Loading HaasoscopeLib.py
('Linux', 'lnx', '5.4.0-45-generic', '#49-Ubuntu SMP Wed Aug 26 13:38:52 UTC 2020', 'x86_64')
('matplotlib backend is', 'TkAgg')
('num main ADC and max10adc bytes for all boards = ', 2048, 'and', 0)
('rate theoretically', 35.95, 'Hz over serial')
('/dev/ttyUSB0', ':', 'USB2.0-Serial', ':', 'USB VID:PID=1A86:7523 LOCATION=1-4')
('connected serial to', '/dev/ttyUSB0', ', timeout', 3.0, 'seconds')
('got firmwareversion', 15, 'for board', 0, 'in', 2.16, 'ms')
('minimum firmwareversion of all boards is', 15)
rolling trigger
('num samples is', 512)
('123 send increment is', 0)
('143 do highres is', True)
('129 trigger time over/under thresh now', 1, 'and usedownsamplefortriggertot is', True)
('serialdelaytimerwait is', 100)
('tell SPI setup: 131 ', 8, 0)
('tell SPI setup: 131 ', 6, 16)
('tell SPI setup: 131 ', 4, 36)
('tell SPI setup: 131 ', 5, 36)
('tell SPI setup: 131 ', 1, 0)
usb2 connection not available
('got uniqueID', '8098227110eec444', 'for board', 0, ', len is now', 1)
('current directory is', '/home/merck/Code/Haasoscope/software')
('reading calibrations for board', 0, ', channels', 0, '-', 4)
('read', 'calib/calib_8098227110eec444.json.txt')
('calib was written using firmware version', 15)
('theboard:', 0, ' num_board:', 1, ' chan:', 0, ' num_chan_per_board:', 4)
('theboard:', 0, ' num_board:', 1, ' chan:', 1, ' num_chan_per_board:', 4)
('theboard:', 0, ' num_board:', 1, ' chan:', 2, ' num_chan_per_board:', 4)
('theboard:', 0, ' num_board:', 1, ' chan:', 3, ' num_chan_per_board:', 4)
('143 do highres is', False)
shut down adcs
bye bye!

closed 2021-04-17 by drandyhaas