drandyhaas/Haasoscope #64: Running in Python 3 on Raspberry Pi

strwxq, 2021-02-21:
Was able to setup the board and run the  HaasoscopeDraw.py in Windows with Python 2,7 succesfully.
Trying to setup and run on Raspberry Pi with Python 3.7
Getting this error

Python 3.7.3 (/usr/bin/python3)
>>> %Run HaasoscopeDraw.py
Traceback (most recent call last):
  File "/home/pi/Desktop/Haasoscope-python3/software/HaasoscopeDraw.py", line 63
    if d.db: print time.time()-d.oldtime,"done with evt",nevents
                      ^
SyntaxError: invalid syntax

Before this I was not able to install PySide2, getting this error:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting PySide2
  Could not find a version that satisfies the requirement PySide2 (from versions: )
No matching distribution found for PySide2

everything else in pip3 install pyserial numpy matplotlib Qt.py scipy pyqtgraph PySide2 command seems to be there 

Not big in Linux here, just following directions so please, if anybody can explain in plain terms what I am doing worng I'd appreciate it.

drandyhaas, 2021-02-21:
Note that you have to use the python3 branch of the Haasoscope git repo...

strwxq, 2021-02-21:
The folder where all the software resided is called Haasoscope-python3 so I assume I am using the python 3 version of the code

drandyhaas, 2021-02-21:
You would have had to checkout the python3 branch. What does
git status
say?

strwxq, 2021-02-21:
I downloaded the zip file and placed the folder on desktop. Did not use git command.

drandyhaas, 2021-02-21:
Ah, OK.
But you're supposed to run 
python HaasoscopeQt.py

strwxq, 2021-02-21:
Just ran it and here's what I am getting
It seems like something is not right with numpy 
I went back and ran sudo apt-get install libatlas-base-dev
The error seems to be the same before and after

 %Run HaasoscopeQt.py
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Desktop/Haasoscope-python3/software/HaasoscopeQt.py", line 6, in <module>
    import numpy as np
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/__init__.py", line 145, in <module>
    from . import core
  File "/home/pi/.local/lib/python3.7/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/usr/bin/python3"
  * The NumPy version is: "1.20.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

>>>

strwxq, 2021-02-21:
The NumPy version is 1.20.1

strwxq, 2021-02-21:
Sorry scratch that.
I ran the sudo apt-get install libatlas-base-dev on a different raspberry pi. I have two of them opened via remote desktop on my desktop. My bad. 
So I ran the sudo apt-get install libatlas-base-dev on the raspberry pi the haasscope board is connected to and this is what I get now from running HaasoscopeQT.py

"
Loading HaasoscopeLibQt.py
posix.uname_result(sysname='Linux', nodename='raspberrypi', release='5.4.51-v7+', version='#1333 SMP Mon Aug 10 16:45:19 BST 2020', machine='armv7l')
On a raspberry pi?
Number of arguments: 1 arguments.
Argument List: ['HaasoscopeQt.py']
num main ADC and max10adc bytes for all boards =  2048 and 0
Construction done
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
rate theoretically 66.58 Hz over serial
/dev/ttyAMA0 : ttyAMA0 : 3f201000.serial
No serial COM port opened!
"

strwxq, 2021-02-21:
Sorry, scratch that too!!! 
My desk is a mess.
The Haas-scope board was connected to the wrong raspberry pi too!!!!! The HaasoscopeQT.py window has opened !!! 
Huge embarrassment here. 
Now I am confused a bit but the graphics look completely different than in Windows. Looks like it's got more options but some of the text in the right window is cutoff

strwxq, 2021-02-21:
It keeps saying this and the image is static although I have a PWM signal connected to channel 0

don't have channel 0 on board 0
getdata asked for 2048 bytes and got 735 from board 0
(131, 0, 0, 0, 0, 0, 0, 0, 0, 0)
don't have channel 0 on board 0
getdata asked for 2048 bytes and got 768 from board 0
(0, 0, 0, 0, 0, 0, 0, 0, 0, 0)
don't have channel 0 on board 0

strwxq, 2021-02-21:
This is how my screen looks like

According to the little screen it should be showing something different

[image]

drandyhaas, 2021-02-21:
The "getdata asked for 2048 bytes and got 73..." is in the faq at the end of the guide..

strwxq, 2021-02-21:
Yep, Saw that.

This is going on a raspberry pi 3 plus. I have to update the usb. Got that.
I will try now to run it on the other raspberry pi which is a 4 and it's got two blue USB's . They should be a at least USB2's

This one is running NumPy 1.16.2 and it should be running at least NumPy 1.16.5. or at least that is the message your program errors out with. I am having a problems updating to that although I ran the same commands on both raspberry pi's .  Weird.

strwxq, 2021-02-22:
Ok. 

After numerous unsuccessful attempts in removing numpy 1.16.2 via prompt commands I was able to remove it using the Add/Remove Software in the GUI. 'Wish I thought of it earlier. With it, some other packages were automatically uninstalled and to be honest I am not sure how useful they were to what I am doing, but I did it anyway. I then installed numpy 1.20 using the same GUI. 

That fixed the problem and the HaasoscopeQT.py is working now correctly, it seems.
Sorry for the confusion between the HaasoscopeQT.py and the HaasoscopeDraw.py. I wasn't paying attention. 

Again, this is a Raspberry Pi 4 and Haas-scope board is plugged in into one of the 2 blue colored USB 3's. My other Raspberry Pi 3 b+ needs an upgraded USB because the 4 onboard USB 2's aren't fast enough.   
 
This is what the Shell output is before the program starts counting and displaying events 
The line starting with "libEGL warning"  and the next two lines are displayed in red, 'not sure what that means, maybe you can clarify. 

Some of the text lines in the right pane are cutoff or look like the text is not fitting within the space allowed and that may have something to do with the warning above or the PySide2 error I was getting earlier.

*****************************************
Loading HaasoscopeLibQt.py
posix.uname_result(sysname='Linux', nodename='raspberrypi', release='5.4.51-v7l+', version='#1333 SMP Mon Aug 10 16:51:40 BST 2020', machine='armv7l')
On a raspberry pi?
Number of arguments: 1 arguments.
Argument List: ['HaasoscopeQt.py']
num main ADC and max10adc bytes for all boards =  2048 and 0
Construction done
**_libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no_**
rate theoretically 66.58 Hz over serial
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=1-1.1
/dev/ttyAMA0 : ttyAMA0 : fe201000.serial
connected serial to /dev/ttyUSB0 , timeout 3.0 seconds
got firmwareversion 16 for board 0 in 2.46 ms
minimum firmwareversion of all boards is 16
rolling trigger
num samples is 512
send increment is 0
do highres is True
trigger time over/under thresh now 1 and usedownsamplefortriggertot is True
serialdelaytimerwait is 0
tell SPI setup: 131  8 0
tell SPI setup: 131  6 16
tell SPI setup: 131  4 36
tell SPI setup: 131  5 36
tell SPI setup: 131  1 0
usb2 connection not available
got uniqueID 8098227110eebf84 for board 0 , len is now 1
current directory is /home/pi/Desktop/Haasoscope-python3/software
reading calibrations for board 0 , channels 0 - 4
No calib file found for board 0 at file calib/calib_8098227110eebf84.json.txt
100 events, 18.14 Hz
200 events, 24.31 Hz
300 events, 24.3 Hz
400 events, 24.19 Hz
****************************************

drandyhaas, 2021-02-22:
Looks good!

closed 2021-02-23 by strwxq