drandyhaas/Haasoscope #70: Firmware upgrade

danielkucera, 2021-12-13:
**What are you trying to do?**
I want to upgrade firmware of the board. Is there any guide/files? I have USB Blaster.
I was getting error when running the client:
```
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=2-1
connected serial to /dev/ttyUSB0 , timeout 0.37014933333333333 seconds
got firmwareversion 0 for board 0 in 100.32 ms
Could not read firmware version - Exiting!
```
so I commented one line:
```
--- a/software/HaasoscopeLibQt.py
+++ b/software/HaasoscopeLibQt.py
@@ -254,7 +254,7 @@ class Haasoscope():
         print("got firmwareversion",firmwareversion,"for board",board,"in",round(tookms,2),"ms")
         if firmwareversion==0 and tookms>50:
             print("Could not read firmware version - Exiting!")
-            sys.exit(-7)
+            #sys.exit(-7)
         return firmwareversion # is 0 if not found (firmware version <5)
     
     def telltickstowait(self):
```
and it runs now:
```
Loading HaasoscopeTrigLibQt.py
Argument List: ['HaasoscopeQt.py']
num main ADC and max10adc bytes for all boards =  2048 and 0
Construction done
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
rate theoretically 35.95 Hz over serial
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=2-1
connected serial to /dev/ttyUSB0 , timeout 0.37014933333333333 seconds
got firmwareversion 0 for board 0 in 100.36 ms
Could not read firmware version - Exiting!
minimum firmwareversion of all boards is 0
rolling trigger
num samples is 512
send increment is 0
do highres is True
trigger time over/under thresh now 1 and usedownsamplefortriggertot is True
serialdelaytimerwait is 100
tell SPI setup: 131  8 0
tell SPI setup: 131  6 16
tell SPI setup: 131  4 27
tell SPI setup: 131  5 27
tell SPI setup: 131  1 0
usb2 connection not available
got uniqueID 80988211106e2028 for board 0 , len is now 1
current directory is /storage/Projects/Haasoscope/software
reading calibrations for board 0 , channels 0 - 4
read calib/calib_80988211106e2028.json.txt
calib was written using unknown firmware version
```

The QT UI works now, but I would prefer to update the FW anyway.
My HW version is v8.81

**What is your OS and python version?**
Linux, Python 3.9.7

**Have you installed the prerequisites?**
Yes

drandyhaas, 2021-12-13:
Instructions for upgrading with new firmware are in the Haasoscope guide.
But new firmware for v8 boards has not been made for some time. The python3 Qt gui is only tested with v9 boards and their associated firmware.

danielkucera, 2021-12-13:
Not the best support I have to say. And I wanted to write integration for glscopeclient... Nevermind, I'll find some other hardware.