drandyhaas/Haasoscope #71: Could not read firmware version - Exiting!

bradkyle, 2022-02-02:
I want to launch HaasoscopeQt.py

OS:Arch linux, python:3.10.2, conda environment

$python ./HaasoscopeQt.py
Loading HaasoscopeTrigLibQt.py
Argument List: ['./HaasoscopeQt.py']
num main ADC and max10adc bytes for all boards =  2048 and 0
Construction done
rate theoretically 35.95 Hz over serial
/dev/ttyUSB0 : USB2.0-Serial : USB VID:PID=1A86:7523 LOCATION=1-1.2.2
connected serial to /dev/ttyUSB0 , timeout 0.37014933333333333 seconds
got firmwareversion 0 for board 0 in 100.34 ms
Could not read firmware version - Exiting!

expected launch haasoscope software.

I have tried flashing latest firmware, updating libraries, changing python env, rebooting computer, resetting haasoscope, unplugging probes, rebooting haasoscope, reinstalling all libraries, changing udev rules.
I also modified the HaasoscopeLibQt.py to rule out the error: however it then throws the error that it cant connect via usb?

Could anyone please advise me how I can resolve or at least troubleshoot this?
Thanks

drandyhaas, 2022-02-02:
What hardware version of the board do you have? v8 or v9?
I assume you are running this from the master branch of the git repository? And it is up to date (git pull)?

bradkyle, 2022-02-09:
Yes:
Board harware version: v9.
Running from master branch.

drandyhaas, 2022-02-09:
You closed the issue. Did you get it to work?

bradkyle, 2022-02-11:
No, the problem persists.

bradkyle, 2022-02-11:
I've tried changing usb cables, usb ports, computers, operating systems. On occasion it launches but the cause therein is not readily verifiable.

drandyhaas, 2022-02-11:
It can also be the soldering of the 50MHz oscillator near the Fpga. Try pressing down on it. Or reflow it if you know how.

bradkyle, 2022-02-11:
Just tried this. The failure generally triggers when a secondary bnc probe is attached//signal collected. I don't know if there exists a causal link just yet.

drandyhaas, 2022-02-11:
Even if the other end of the probe is not attached to anything? That would be weird. 
I could imagine big problems if the ground clip of the attached probe is put on something with a voltage significantly different from the ground of the USB connection...

closed 2022-12-02 by drandyhaas