drandyhaas/Haasoscope #83: Strange behavior when try to increase functionality

bnusbick, 2025-01-28:
Hi,

As I mentioned I would do previously, I have  rewritten most of the FPGA and PC code for my Haasoscope, and am rather pleased with the result.

What I would like to do now is increase the voltage range that can be fed into m V9.01 Haasoscope.  I know I can use the 10x mode to increase the range, but that is too much, and thus I would get very few bits for a signal that changed from say 0-5 volts.  My idea is to use the same technique that x10 uses, namely putting a large resistor in parallel with a capacitor before the scope 1Mohm resistance.

For my initial experiments, I have added 1.330 M ohm resistance, and thus would expect the voltage measured would be 1/2.33 of the reading that would otherwise be read.  After adjusting the zero offset, my Haasocope reads 128 when I short the probe ground to tip, 0 when I apply 4 volts, and 32 when I apply 3 volts, so it appears the maximum voltage is 4 volts.  With the additional 1.33 M ohm resistance, I would expect the reading to be about 128 - (3/4) * 128 * (1/2.33) = 41.2, but it reads 72.

Also, if I short the probe ground to the end of the 1.33 M ohm connected to the probe tip, it measures 113 instead of the expected 128.

I am not much of a hardware person (but I am try to be more so), so I am not able to understand this behavior from looking at the schematic.  Could you explain why I see what I do?  If there is some additional calculation I need to perform to adjust for the addition of 1.33M ohm (or other values to get larger ranges), I could easily adjust the PC code so it would display correctly.

As a sanity check, I applied the same additional 1.33 M ohm to the tip of a probe connected to a Tektronix 2230, and the voltage I measured when applying 3 volts was 1.3 volts  (3 * 1 M ohm / 2.33 M ohm) as expected, and 0 volts when I connected the probe tip to the 1.33 M ohm additional resistance that I added to the the scope tip.

Thanks in advance for any help you can provide.

drandyhaas, 2025-01-28:
Sounds like a good project for learning!
I'm a bit unclear what you're trying to do. You want not 10x gain, but say, 3x gain? The gain is controlled by 1MOhm RGAIN2_1 and RGAIN4_1, in parallel with the 3-10pF variable cap. You would adjust those values to change the gain.

bnusbick, 2025-01-28:
I am trying to create a 2.3x gain without modifying the board.  Looking on the web, scope probes add a 9M ohm resistor between the probe tip and the 1 M ohm resistor to create 10x gain. I was hoping adding 1.3 M ohm at the probe tip, and then connecting to a gripper that I attach to this would give 2.3x gain.

bnusbick, 2025-01-29:
If I short the probe ground lead to the end of the additional 1.33 M ohm resistance I added to the probe tip, and then offset that reading to 0, I get the proper behavior, i.e. the readings are 1/2.33 of what they would be without the additional 1.33 M ohm resistance, and when I feed 9.4 volts in, the A/D converter sends 0.

closed 2025-01-30 by bnusbick